pikvm/pikvm #1402: Linux kernel console over USB serial through OTG

QRPp, 2024-09-17:
(A need similar to #945 but deeper, not satisfied by CDC ADM port available now through OTG.)

_Problem:_ need to **log console output of an amd64 Linux host, most importantly the kernel crash dumps etc.**

_Preferred solution:_ do that **through a PiKVM's OTG-connected serial port.**  The currently [configurable](https://docs.pikvm.org/usb_serial/) `ttyACM0` port is handled by the `cdc_acm` Linux kernel module, which doesn't use `usbserial`, thus can't be enabled via a `console=tty...` Linux cmdline argument at least on x86/amd64 arch.

_Alternatives:_
- Constantly capturing the VGA/HDMI console _video stream:_ _high overhead, hassle to copy/paste_ and still easily _misses console output_ when produced fast enough.
- An _extra USB serial adapter_ on the Linux host hooked up:
  - to _PiKVM's UART_ (GPIO 14/15): works but _prevents doing the same to PiKVM in turn,_ which is handy for the overall robustness, plus means _extra cabling and hardware._
  - to an _extra USB serial adapter on PiKVM:_ works but means _even more extra cabling and hardware,_ at the very least [something like this](https://www.aliexpress.com/item/4000544494503.html).

_Questions:_
- _Is it technically possible_ for PiKVM to emulate a USB OTG serial port that'd work via `usbserial` module in Linux like USB serial adapters do?
- If yes, _is it on the plans?_

Cheers!

mdevaev, 2024-09-17:
It's not possible. OTG subsystem is working with some other module set than `usbserial`.

QRPp, 2024-09-18:
Thought it wasn't possible, thanks for confirming, alas.  Well, after all, it's probably more productive to consider solving this from the other end, i.e., making `cdc_acm` ports usable as consoles in the kernel.  Thanks for all the good work on PiKVM!

closed 2024-09-18 by QRPp