pikvm/pikvm #1403: Add documentation about ch9329 HID

manuvaldi, 2024-09-21:
**Is your feature request related to a problem? Please describe.**
This option is not documented but it seems to be supported.

**Describe the solution you'd like**
Add documentation and examples to configure this HID in https://docs.pikvm.org/

**Describe alternatives you've considered**
--

**Additional context**
Documentation in discord?
[image]

mdevaev, 2024-09-21:
It's strongly not recommended to use:

* No absolute/relative configuration in protocol;
* No protocol synchronisation (can just hang);
* Occupies the UART that is used for the linux console.

closed 2024-09-21 by mdevaev

manuvaldi, 2024-09-22:
I think it's not a problem add documentation and explain this is only for experiments and it's not supported and will not.

mdevaev, 2024-09-22:
Why document a non-recommended unsupported solution if there is a better alternative?

manuvaldi, 2024-09-22:
To be honest, the better alternative for rpi 2/3 is assembling, and this is not for all people. Other open source projects accept ch9329 as solution. I'm agree with you people do not stop just for a warning, but this is something that should be documented and warned. I'm in the open source enterprise world since many years ago and the customers understand perfectly what is supported and unsupported, and the associated risks. 

For the enterprise world and people don't like risk v3/v4 even v2 is the solution. But there are a big bunch of people who like and use pikvm as a open source project in their labs/experiments (and with some old rpi2/3 and low budget) and we understand the risks but also the advantaged of something simple with no circuits or cables... 

I encourage you to reopen the issue.

mdevaev, 2024-09-22:
> To be honest, the better alternative for rpi 2/3 is assembling, and this is not for all people.

Pico HID has a UART mode to be a complete analog of the ch9329. That is, it can be used with the same two UART wires, plus GND and VCC. Even from the point of view of the complexity of the assembly, there is no point in using ch9329, given that it works worse than Pico in all other aspects.

> Other open source projects accept ch9329 as solution.

This does not mean that I should automatically agree with their decision. Popularity does not mean that this thing is good. What PiKVM provides should work reliably, because PiKVM is positioned as an emergency access device. The listed disadvantages of the protocol make ch9329 unreliable.

> I'm in the open source enterprise world since many years ago

Then you should definitely understand me. In addition to technical problems, a wide variety of assemblies and circuits creates problems for the support team. I spent a lot of time in order to unify the DIY designs. But you suggest that I add and document another option so that it leads to senseless confusion again.

If the user wants to use ch9329, then the minimum necessary skill for this is the use of search. I suppose you wanted a ch9329 and got it in Discord, so you didn't have a problem with it. Otherwise, it is better for the user to follow the recommended way.

manuvaldi, 2024-09-23:
Rpi 2/3 users would like a solution without assembly like rpi4, it's just that. There are ch9329 based cables already ready to plug-and-play. The point of the asembling + cables + chip board is not a "clean" solution. The clean solution is to buy ready-to-use usb cables and an usb capture, plug and play with pikvm. Assembling stop the adoption for many users. Chip boards, wires, aseembly boards, wiring diagrams, etc (all without any protection case) scare for no electronic aware people (like me).  I don't feel confortable, and like me there are many users. This is my point, sorry.

NiKiZe, 2024-09-23:
> Rpi 2/3 users would like a solution without assembly like rpi4, it's just that. There are ch9329 based cables already ready to plug-and-play. The point of the asembling + cables + chip board is not [...]

If so, buy a full PiKVM, not some non working 3rd party cable, if you really want that, use the search function.

One other solution is to document existence of non recommended variants on the web as well, and then link to GitHub search results for more info.

manuvaldi, 2024-09-23:
Full PiKVM is expensive (mini in EU ~300$), and  I would say out-of-market for KVM-IP solutions. If you already have an rpi 2/3 and no idea of assembly, you buy 5$ usb capture and 10$ usb ch9329 cable, and for 15$ you can use pikvm. This is so powerful that idea... it helps in the pikvm adoption.

The solution you propose @NiKiZe with a link to github/discord would be great from my point of view.

mdevaev, 2024-09-23:
> Rpi 2/3 users would like a solution without assembly like rpi4, it's just that. There are ch9329 based cables already ready to plug-and-play. The point of the asembling + cables + chip board is not a "clean" solution.

It's clean if you just follow the instructions. You don't need to be able to read schematics, it is enough to connect the wires as in the picture. The difference is only in the number of wires, and this is a reasonable price for eliminating all the listed problems of the ch9329.

> I don't feel confortable, and like me there are many users. This is my point, sorry.

When you put it together, you'll realize that it's not scary at all. And we also help with the assembly in Discord. And you can make any case you want. Well, it's DIY. Users are constantly making this.

> Full PiKVM is expensive (mini in EU ~300$), and I would say out-of-market for KVM-IP solutions. 

You can compare the price with solutions from Lantronix and ATEN. At the same time, PiKVM V3/V4 gives much more features. Just let's compare the prices of new products honestly, and not the price of decommissioned equipment from ebay.

> If you already have an rpi 2/3 and no idea of assembly

Just read a [simple instructions](https://docs.pikvm.org/v1), buy all listed parts and connect it together.

But if we talk about your specific point, then your problem has already been solved, you found it in Discord. Meanwhile, if you act against the instructions, then you remain in the category of self support with regard to hardware. You are trying to persuade me that I should accept this as an option, which (regardless of the number of warnings) will make me responsible for the stability of the solution I'm giving. I have already explained that the ch9329 is a huge pile of problems, and generally a bad chip. The stability of the Pico solution depends on my code and I am confident in it. I have no confidence about ch9329. A poorly working solution will negatively affect the reputation of the entire project.

manuvaldi, 2024-09-23:
I get it. Open sourcing while simultaneously selling ad hoc hardware establishes limits on DIY solutions. Not all solutions, nor the unlimited ideas from people, can be supported. Understood guys

Enterprise open-source companies take open-source projects from upstream and then build a downstream version that is supported and limited. This would lead you to have one fully open-source PiKVM project and another enterprise PiKVM project, which, to be honest, is complicated.

Related with that I found someone is layering pikvm and supporting ch9329 https://one-kvm.mofeng.run/ch9329_hid/ . At least they mention pikvm source. XD

Thank you for your support and understanding, guys. I’m leaving the topic.

mdevaev, 2024-09-23:
> I get it. Open sourcing while simultaneously selling ad hoc hardware establishes limits on DIY solutions. Not all solutions, nor the unlimited ideas from people, can be supported. Understood guys

You just ignore my technical and organizational rationale and came up with your own answers. Before Pico HID was introduced, there was the [Arduino HID](https://docs.pikvm.org/arduino_hid/), which was much more complex. Pico HID appeared **after** we started selling our hardware, and precisely in order to simplify the DIY path. So your assumption has nothing to do with reality. Despite the fact that we sell hardware, we continue to improve and support DIY.

> This would lead you to have one fully open-source PiKVM project and another enterprise PiKVM project, which, to be honest, is complicated.

Our hardware gives you capabilities that are not available in DIY builds due to excessive complexity or hardware limitations of components. Some things like 1080p@60Hz video with H.264 just can't be done any other way.

> Related with that I found someone is layering pikvm and supporting ch9329

No problem, you can use that fork.