pikvm/pikvm #1405: LED state reported only one time

tomiko23lol, 2024-09-23:
Hello I configured view that way that my header contains LEDs and when I click on this header it should be expanded with the same LED states, just with buttons to control it next to it.
Problem is that LED state is displayed only once. When I have it in Header, it is not reported in table. As both LED displays are the same thing, I would suspect that both should show the same thing at the same time ...

[image]

Here is my configuration:

`
[image]
`

**Expected behavior**
I would expect that if I put in status of the same led 10 times, it will works 10 times the same way.

**PiKVM info:**
  - Raspberry Pi zero 2 w
  - 8 GPIO inputs (for LEDs) + 8 GPIO outputs (as button controls)
  - Video capture CSI bridge C790

mdevaev, 2024-10-01:
Confirmed. I'll fix it soon.

mdevaev, 2024-10-02:
Fixed. Please update OS.

closed 2024-10-02 by mdevaev

tomiko23lol, 2024-10-04:
It works now, thank You very much :) 

[image]

mdevaev, 2024-10-04:
:ok_hand: