pikvm/pikvm #1406: Keys overlap on virtual keyboard when running in high DPI mode on Chromium-based browsers

jernejs, 2024-09-26:
**Describe the bug**
Some of the keys on the Virtual Keyboard rearrange and overlap when the keyboard is moved, and computer is running in high-DPI mode (specifically, I'm running Windows at 225%, though I've reproduced it at 175%, too). Seems to only happen in Chromium-based browsers (reproduced in Vivaldi, Edge and Chromium-based webpage plugin in RoyalTS; does not happen in Firefox).

**To Reproduce**
Steps to reproduce the behavior, like:
1. Set Windows DPI to at least 175%
2. Go to PiKVM, click System and Show keyboard
3. Try to move the virtual keyboard
4. Look at the Ins/Del/PgUp/PgDn keys

**Expected behavior**
Keys don't move around

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 23H2
 - Browser: Vivaldi 6.10.3468.4, Edge 129.0.2792.52

**PiKVM info:**
  - Raspberry Pi board version: RPi4+PiKVM v3, PiKVM v4
  - Video capture type: CSI bridge
  - KVMD version: kvmd 4.16-1
  - uStreamer version: ustreamer 6.16-1
  - Linux kernel: `Linux pikvm 6.6.45-7-rpi #1 SMP Mon Sep  2 09:16:27 UTC 2024 armv7l GNU/Linux`

AndrewPaglusch, 2024-10-30:
I'm experiencing the same issue here. It's not Chromium specific -- I'm using Firefox.

**Desktop:**
OS: macOS Sequoia 15.0.1
Browser: Firefox
Version: 132.0 (aarch64)

**PiKVM info:**
Raspberry Pi board version: CM4 + PiKVM V4 Mini
Video capture type: CSI bridge
KVMD version: 4.20-1
uStreamer version: 6.16-1
Linux kernel: Linux pikvm 6.6.45-8-rpi #1 SMP Wed Oct  2 19:56:38 UTC 2024 armv7l GNU/Linux

mattico, 2025-03-11:
Another data point: I'm having the issue with Firefox on Windows running at 150% DPI scaling.

mdevaev, 2025-05-13:
Found a problem. The fix will be available in the next release.

mdevaev, 2025-05-18:
Released in 4.72, please update with `pikvm-update`.

closed 2025-05-18 by mdevaev