pikvm/pikvm #1408: Please add more LED colors possibly usable for buttons

tomiko23lol, 2024-09-30:
Hello as far as documentation goes, there are only 3 colors available for LEDs.

[image]

Please give us more color options, ideally following:

RED
GREEN
DARK_GREEN
BLUE
CYAN
PURPLE
PINK
ORANGE
YELLOW
WHITE

more than 8 colors will allow us to distinguish LED colors in case of up to 8 ports on KVM switches etc.

If possible it would be nice to give us option to use same color config for buttons as well ( for example to distinguish restart buttons with red color. Automatically select text black or white depending on what is more suitable for particular color to provide good contrast )

Thank You in advance.

tomiko23lol, 2024-09-30:
Or as alternative user specified color code in HEX or RGB format

mdevaev, 2024-11-26:
Hello. I've added `blue`, `cyan`, `magenta`, `pink` and `white`. These are distinguishable colors, I can't add completely custom RGB.

closed 2024-11-26 by mdevaev

tomiko23lol, 2024-11-26:
Thank you very much for quick change. This will help a lot. 😀