pikvm/pikvm #1409: Custom Resolution

chchia, 2024-10-01:
**Is your feature request related to a problem? Please describe.**
I have a machine which running with non-standard custom resolution, at 1056x1116. i tried to change it in main.yaml but it doesnt seem to support it. i am using DIYv2 with USB capture dongle.

**Describe the solution you'd like**
supporting a custom resolution with correct display ratio will be appreciated.

[image]

[image]

chchia, 2024-10-01:
thank you, although i can now set custom resolution, however the aspect ratio still cant be changed.

[image]

mdevaev, 2024-10-01:
Try to calculate the correct screen ratio and use it as a fake resolution.

mdevaev, 2024-10-04:
Sup?

mdevaev, 2024-10-06:
Closing due no feedback.

closed 2024-10-06 by mdevaev