pikvm/pikvm #1415: Does not autoconnect to open wifi on boot

reukiodo, 2024-10-19:
**Describe the bug**
following https://docs.pikvm.org/on_boot_config/#setting-up-wi-fi I added `WIFI_ESSID='mynet'` to `\boot\pikvm.txt` but it never connected to wifi
following https://github.com/pikvm/pikvm/issues/185#issuecomment-779330909 to disable usb, logged into the console and used `iw dev wlan0 connect home` to connect to wifi without issue

**To Reproduce**
Steps to reproduce the behavior, like:
1. add `WIFI_ESSID='mynet'` to `\boot\pikvm.txt`
2. wait 1hr, no connection
3. disable usb by commenting `dwc2` line in `/boot/config.txt`
4. log into console and validate wifi connection

**Expected behavior**
It should automatically configure wifi to always connect to open WiFi network.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Arch?
 - Browser n/a
 - Version Linux version 6.6.45-7-rpi (alarm@fb49127d3e4f) (distcc 3.4 armv7l-unknown-linux-gnueabihf, GNU ld (GNU Binutils) 2.42.0) #1 SMP Mon Sep  2 09:16:27 UTC 2024
 - VNC client n/a

**PiKVM info:**
  - Raspberry Pi board version RPi0 2
  - PiKVM platform v2-hdmi
  - Video capture type n/a
  - KVMD version: 4.10-1
  - uStreamer version: 6.14-1
  - Linux kernel: Linux pikvm 6.6.45-7-rpi #1 SMP Mon Sep  2 09:16:27 UTC 2024 armv7l GNU/Linux

**Additional context**
Haven't even gotten to connecting capture card or anything like that, WiFi needs to connect more reliably first.

reukiodo, 2024-10-19:
After a LOT of digging, I finally found https://ricostacruz.com/til/arch-linux-wifi-using-netctl which had the answers to set up netctl to automatically connect to wifi on each boot. Good workaround.

Still a bug as it should work as according to documentation though.

mdevaev, 2024-11-20:
Fixed. It will be available in the next release.

closed 2024-11-20 by mdevaev

reukiodo, 2024-11-21:
Thanks!

Is there any easier way to determine the release that will include this than from comparing the date of the latest file in https://files.pikvm.org/images/ ?

mdevaev, 2024-11-22:
No, sorry.