pikvm/pikvm #1420: When connecting using bvnc clipboard is pasted upon connect every time

Raptor-TK, 2024-10-30:
**Describe the bug**
When connecting using bvnc, clipboard from android is pasted upon connect every time.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Have something in android clipboard
2.  Connect to pikvm vnc server using bvnc on android
3. The clipboard is automatically pasted into the pikvm
4. Pastes every time. Very annoying.

**Expected behavior**
The clipboard should not be pasted upon connect.

**VNC client phone:**
 - OS: Android 14
 - Browser N/A
 - Version [e.g. 22] android 14
 - VNC client (if used) bvnc

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4] RPi 4
  - PiKVM platform [e.g. v2-hdmi] DIY PiKVM V2 Platform hdmi to usb dongle version
  - Video capture type [e.g. CSI bridge] hdmi to usb
  - KVMD version: `pacman -Q | grep kvmd` 4.10
  - uStreamer version: `pacman -Q | grep ustreamer` 6.14
  - Linux kernel: `uname -a` Linux pikvm 6.6.45-7-rpi #1 SMP Mon Sep  2 09:16:27 UTC 2024 armv7l GNU/Linux

**Additional context**
KVMD: 4.10
Streamer: 6.14 (ustreamer)
HAS_PDEATHSIG: Yes
WITH_GPIO: Yes
WITH_PTHREAD_NP: Yes
WITH_SETPROCTITLE: Yes
WITH_SYSTEMD: Yes
Linux kernel:
Machine: armv7l
Release: 6.6.45-7-rpi
Version: #1 SMP Mon Sep 2 09:16:27 UTC 2024

Raptor-TK, 2024-10-30:
#284 #926 are related issues.

Raptor-TK, 2024-10-31:
One workaround is to use an app like the following to clear the clipboard in android before connecting to the PiKVM.

https://github.com/amnesica/ClearClipboard

mdevaev, 2024-11-02:
I suppose this is a problem on the client side. PiKVM receives a clipboard insertion signal from them and simply executes this command.

mdevaev, 2024-11-02:
Please attach `journalctl -u kvmd-vnc` log. If this event arrives immediately after the connection, I can add some delay during which clipboard events will be ignored.

Raptor-TK, 2024-11-02:
I reproduced the issue copying the word "Test" before connecting to the PiKVM several times.
In the log it prints:
```
Nov 02 15:20:39 pikvm kvmd-vnc[502]: kvmd.apps.vnc.server              INFO --- [::ffff:192.168.0.51]:46874 [kvmd]: Connected to KVMD websocket
Nov 02 15:20:39 pikvm kvmd-vnc[502]: kvmd.apps.vnc.server              INFO --- [::ffff:192.168.0.51]:46874 [streamer]: Using preferred MemsinkStreamerClient(JPEG)
Nov 02 15:20:39 pikvm kvmd-vnc[502]: kvmd.apps.vnc.server              INFO --- [::ffff:192.168.0.51]:46874 [streamer]: Streaming ...
Nov 02 15:20:39 pikvm kvmd-vnc[502]: kvmd.apps.vnc.server              INFO --- [::ffff:192.168.0.51]:46874 **[main]: Printing 4 characters ...**
``` 
[kvmd-vnc.log](https://github.com/user-attachments/files/17607458/kvmd-vnc.log)

mdevaev, 2024-11-02:
Try this hack. Patch your file `/usr/lib/python3.12/site-packages/kvmd/apps/vnc/rfb/__init__.py` like this:
```diff
@ kvmd/apps/vnc/rfb/__init__.py:25 @

import asyncio
import ssl
import time

from typing import Callable
from typing import Coroutine
@ kvmd/apps/vnc/rfb/__init__.py:96 @ class RfbClient(RfbClientStream):  # pylint: disable=too-many-instance-attribute

        self.__lock = asyncio.Lock()

        self.__connect_ts = 0.0

    # =====

    async def _run(self, **coros: Coroutine) -> None:
@ kvmd/apps/vnc/rfb/__init__.py:420 @ class RfbClient(RfbClientStream):  # pylint: disable=too-many-instance-attribute
    # =====

    async def __main_loop(self) -> None:
        self.__connect_ts = time.monotonic()
        handlers = {
            0: self.__handle_set_pixel_format,
            2: self.__handle_set_encodings,
@ kvmd/apps/vnc/rfb/__init__.py:506 @ class RfbClient(RfbClientStream):  # pylint: disable=too-many-instance-attribute
    async def __handle_client_cut_text(self) -> None:
        length = (await self._read_struct("cut text length", "xxx L"))[0]
        text = await self._read_text("cut text data", length)
        await self._on_cut_event(text)
        if time.monotonic() > self.__connect_ts + 3:
            await self._on_cut_event(text)

    async def __handle_enable_cont_updates(self) -> None:
        enabled = bool((await self._read_struct("enabled ContUpdates", "B HH HH"))[0])
```
If you do not know how to do this, I will send you the whole file.

mdevaev, 2024-11-02:
Updated ^^^

Raptor-TK, 2024-11-03:
Thanks I'll try this and update you when I've tested it.

Raptor-TK, 2024-11-04:
It works!
Tested it several times with both AVNC and bVNC clients in android.

Thank you for resolving this issue!

closed 2024-11-04 by mdevaev

Raptor-TK, 2024-11-05:
Actually,  I jumped the gun.
I found another scenario it happens.

When VNC Client is running but was running in background but not disconnected.
If you switch focus back to it, the clipboard contents is pasted still.

In this case I'm not sure if there is an easy solution.
But at least the situation of it pasting the clipboard on the initial connect is not happening.

mdevaev, 2024-11-05:
There's no way I can fix it. VNC is not really designed for something like PiKVM, and clients are trying to synchronize clipboard contents with the host. But PiKVM doesn't have a clipboard, only keyboard access, so it tries to type this.

Raptor-TK, 2024-11-05:
Thanks for the hint there that this is likely something that needs solving on the client side.

I checked in AVNC client for android and there is an option to disable clipboard sync that solves this issue.
In bVNC on android, there is no option to disable clipboard sync.