pikvm/pikvm #1426: Nit regarding the PiKVM V2 DIY ATX schematic

henryptung, 2024-11-21:
The design does function as pictured, but as a suggestion, the GPIO pins should probably have protection resistors on them (the inputs in particular). If someone accidentally installs the wrong SD card during setup or maintenance, they could set GPIO24 to output-low and burn out their Pi with a short (don't ask me how I know this).

mdevaev, 2024-11-23:
Hi. I don't think this is necessary, in all the years we have not encountered similar cases from users. It will be enough to do with a warning that the user should carefully check the scheme before powering on.

mdevaev, 2024-11-24:
Added a tip.

closed 2024-11-24 by mdevaev

henryptung, 2024-11-27:
Respectfully, the issue was a software one (booting up with the wrong SD card), not a hardware one (circuit was correct); but thank you for taking a look anyway.

mdevaev, 2024-11-27:
I share your opinion about some protection. But adding even two resistors to the circuit will complicate it for the end user, who is not very experienced at electronics. The zero statistics of such failures over the entire lifetime of the project shows that we can neglect these cases for the sake of simplifying the scheme.

mdevaev, 2024-11-28:
PS: As for the incorrect image, we describe as clearly as possible which image should be chosen. I don't think anything can be improved here.

I forgot to write about it, please don't think that I'm reading your messages inattentively :)

mdevaev, 2024-12-05:
I didn't mean to deceive or offend you and I definitely believe you. I've encountered a lot of weird problems, such as silicon defects in the Raspberry CPU or dying RAM. So you didn't surprise me with the ability to kill Raspberry through misconfigured GPIOs.

I'm just guided by the logic of "this is good enough". A lot of errors can lead to hardware failure, and it is impossible to protect against every user mistake. Therefore, we protect the scheme only from those errors that can actually occur frequently. For example, we could use transistors instead of optocouplers in ATX, but this would impose on the user the need to check the polarity of the connection to the motherboard.

Adding even one part increases the complexity of the scheme on the breadboard and increases the likelihood of making a mistake during assembly. At least that's how I see it. We give the users a very simple scheme and warn them that they should be careful, this is a compromise for the sake of convenience.