pikvm/pikvm #1433: Remmina VNC Plugin Copy/Paste Issues

wesnm, 2024-12-03:
**Describe the bug**
_Note: I know that VNC over Remmina is not officially supported, but the interface is so, so much better than tigervnc that I have to use it to be productive. The PIKVM VNC server tickles what must be several bugs in the VNC plugin, as it periodically crashes Remmina as well. I can live with crashes but this one is really annoying._

When the PIKVM VNC window is open, and Remmina is set to sync the clipboard, copying anything to the clipboard on the client results in it being pasted into the PIKVM window. This does not happen with tigervnc (obviously), and it doesn't happen with Remmina connected to other VNC servers.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Set up PIKVM to be compatible with Remmina's VNC plugin
2. Add the VNC connection to Remmina, ensuring that "disable clipboard sync" is NOT enabled
3. Connect to VNC server. Open an application that accepts text input, such as notepad.
4. Minimize the VNC window
5. Copy text to the clipboard in the client environment
6. Switch back to the VNC session to see the text being typed/pasted into the session.

**Expected behavior**
Text to be synchronized to the client clipboard, not typed.

**Screenshots**
None

**Desktop (please complete the following information):**
 - OS: FreeBSD 14-STABLE
 - Browser: N/A
 - Version: N/A
 - VNC client: Remmina

**PiKVM info:**
  - PIKVM 4+
  - KVMD version:  `kvmd 4.22-1`
  - uStreamer version: `ustreamer 6.17-1`
  - Linux kernel: `Linux pikvm 6.6.45-8-rpi`

mdevaev, 2024-12-03:
> Text to be synchronized to the client clipboard, not typed.

PiKVM doesn't have any clipboard because it emulates a keyboard. It is impossible to transfer your data anywhere except to print it. VNC is not designed to work with something like this at all, and all I can give you is a typing hack.

> Add the VNC connection to Remmina, ensuring that "disable clipboard sync" is NOT enabled

So the solution is enabling this option.

wesnm, 2024-12-03:
So the typing is the expected behavior. I must not be trying hard enough to replicate it in tigervnc, even with clipboard sync enabled, I swear I did not encounter this.

Would it be possible keep the clipboard and intercept a slightly different key combination like win+v, ctrl-shift-v, or ctrl-alt-v and use that to type the clipboard, while still passing ctrl-v as normal.

Zetaphor, 2025-02-26:
>     Set up PIKVM to be compatible with Remmina's VNC plugin

Can you please elaborate on how you accomplished this? I really need Remmina as TigerVNC doesn't support scaling, and I'm on a 4K display

mdevaev, 2025-05-07:
In the next release, to paste over PiKVM you will nee to use magic keys `LeftAlt,LeftAlt,P` quickly pressed. It solves this problem.

mdevaev, 2025-05-18:
Released in 4.72, please update with `pikvm-update`.

closed 2025-05-18 by mdevaev