pikvm/pikvm #1435: Documentation page for ATX control board has a cybersquatted link

GCustom, 2024-12-06:
**Describe the bug**
A clear and concise description of what the bug is.
on https://docs.pikvm.org/atx_board/ the link for "7 Connect the ATX board to PiKVM using a [straight Ethernet cable with 8 wires](https://www.home-network-help.com/straight.html)." the link is being cybersquatted instead leading to a dot kr page written in Thai, it appears to be a gambling site.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to 'https://docs.pikvm.org/atx_board/'
2. Click on 'straight Ethernet cable with 8 wires'
3. wrong page appears

**Expected behavior**
detailed instructions or technical information about straight Ethernet cables appears

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [MacOS 15.1.1]
 - Browser [Edge]
 - Version [131.0.2903.70 (Official build) (arm64)]
 - VNC client (N/A)

**PiKVM info:**
  - Raspberry Pi board version [PiKVM v4 Mini]
  - PiKVM platform [N/A]
  - Video capture type [N/A]
  - KVMD version: `pacman -Q | grep kvmd` N/A
  - uStreamer version: `pacman -Q | grep ustreamer` N/A
  - Linux kernel: `uname -a` N/A

**Additional context**
None

mdevaev, 2024-12-06:
Thank you for reporting. Fixed.

closed 2024-12-06 by mdevaev

GCustom, 2024-12-07:
looks great now