pikvm/pikvm #1436: Add support the ethernet usb hub hat to the raspberry pi zero w 2 version 2 config

DennisOtt-git-hub, 2024-12-10:
**Is your feature request related to a problem? Please describe.**
The problem this new addition would solve is the lack of a control usb port on the Rpi Zero W 2 for controlling multiport switches and provide an ethernet port for stable wired connectivity.

**Describe the solution you'd like**
Include support on the R Pi Zero W config for the widely available ethernet/usb hub hat.  This would provide a way for this configuration to have the control usb port needed to control a switch and a wired ethernet capability.  

**Describe alternatives you've considered**
The only alternative seems to be to use an alternate platform implementing a larger device at higher cost when it's probably not really needed.

mdevaev, 2024-12-10:
There is no technical possibility for this. Zero has only one USB controller, and if you connect such a HAT, a Hub will appear there, which will exclude the possibility of emulating a USB gadget.

closed 2024-12-10 by mdevaev

HritwikSinghal, 2025-07-02:
@mdevaev can we use picoHID for zero2w with an ethernet HAT for getting both ethernet and keyboard emulation?

mdevaev, 2025-07-02:
@HritwikSinghal yes, it's possible.

nixielectra, 2025-07-29:
Hi @mdevaev sorry for tagging you again here, I'm doing some search on Zero2W. so this [HAT](https://www.aliexpress.com/item/1005009065785010.html?spm=a2g0o.order_list.order_list_main.10.78521802mjVI3L) also will not work?