pikvm/pikvm #1440: [Feature Request] H.264 over alternate transport

computergeek125, 2024-12-12:
**Is your feature request related to a problem? Please describe.**
I'm a recent purchaser of a PiKVM V4 mini (which I love and works fully in my local network), but I have an issue with WebRTC in one specific external location.  This location has a firewall (which I do not control) that does not properly support WebRTC and other similar UDP hole-punching technology (such as Parsec) - fixing WebRTC will likely not be possible.  From my research probing Parsec on this firewall with Wireshark and `tcpdump`, it scrambles the source port number arbitrarily which causes STUN connections to be unable to determine their source port accurately.  PiKVM MJPEG and Guacamole in this location work perfectly (terminated at the same firewall on my side).

I'd still like to use H.264 encoding w/ audio support, but to get that to work in this location I believe the server would need to use a transport other than WebRTC to ship the video over.  I recognize that this will likely incur extra latency compared to a pure WebRTC implementation.

**Describe the solution you'd like**
I'm looking for a solution that runs on a protocol that does not require hole punching and can continue operating in a standard browser (i.e. - not using a gateway/VPN program running on localhost).  I believe something like Websocket may be appropriate, but I'll defer to you.   

**Describe alternatives you've considered**
N/A

**Additional context**
I'm willing to test this feature and help with some implementation - but my dev experience is limited to python and some javascript (web).  When I was investigating this issue independently prior to opening this feature request, I was able to trace the potential code paths to this through both kvmd and ustreamer.  I have some familiarity with C (but not a lot anymore), so updating ustreamer may be a bit over my head.

mdevaev, 2024-12-12:
Hello.

There is no any way to get H.264 stream to the browser except WebRTC. Technically, you can actually transmit data in any way, including WebSocket, but there is no way on the receiving side to decode this stream in the browser. You will either have to have an H.264 decoder on JS (slow and CPU-intensive), or repackage the video into a container (introduces significant latency). WebRTC is now the only way to receive H.264 in real time.

You can try using VNC if it is possible to punch at least one hole. TigerVNC supports H.264: https://docs.pikvm.org/vnc

computergeek125, 2024-12-13:
Thanks for responding!  I'll keep hunting.

This _does_ actually explain something I've experienced on Raritan KVMs - they tend to spool the client's fan, and I think I understand why now :P

computergeek125, 2024-12-13:
As an addendum - if I find a solution and have to make code changes to get this to work, would you be interested in the result (i.e. merge request) or is this in the territory of unsupported configurations? My intent is to get this capable of running on stock PiKVM hardware for what it's worth.

mdevaev, 2024-12-13:
If you find a way to quickly transfer H.264 and decode it using native browser code without WebRTC, I will be happy to consider it and add it to the code if possible.

mdevaev, 2024-12-13:
I just remembered that there is a promising way to use WebCodecs, but I haven't gotten around to checking how well it works yet. If you're interested, try some of this.

https://github.com/pikvm/pikvm/issues/784#issuecomment-1890266530

mdevaev, 2024-12-14:
I made quick prototype and it's working. I will add this with medium priority.

mdevaev, 2024-12-25:
Done: https://www.reddit.com/r/pikvm/comments/1hlwkzn/kvmd_429_with_http_h264_streaming/

closed 2024-12-25 by mdevaev