pikvm/pikvm #1441: HP Prodesk 400

alexisfrjp, 2024-12-16:
The HP Prodesk 400 G2 doesn't detect the USB, either keyboard, mouse or mass storage until it boots the OS.
It makes it impossible to go to the BIOS settings or to select the boot option.

What could it be?

mdevaev, 2024-12-16:
HP BIOSes are buggy. You can use a tricky workaround:
* Enable dynamic USB configuration feature: https://docs.pikvm.org/usb_dynamic/
* Disable mass storage using it
* Go to BIOS, make all things, reboot
* Enable mass storage again if you need, boot your laptop from it

alexisfrjp, 2024-12-16:
Thanks! You replied while I was editing my question. I tried disabling the mass storage and I think it worked.
I should focus on the display now.

mdevaev, 2024-12-16:
Some laptops simply not send the BIOS image to an external monitor. There's nothing you can do about it.

alexisfrjp, 2024-12-17:
> Some laptops simply not send the BIOS image to an external monitor. There's nothing you can do about it.

It isn't a laptop, it's a micro desktop like a NUC. 
I can see the BIOS image when it's connected to an external screen.

mdevaev, 2024-12-26:
Which PiKVM device do you use?

alexisfrjp, 2025-01-01:
> Which PiKVM device do you use?

V3 HAT

mdevaev, 2025-01-04:
I looked at the specifications of your device. Where exactly do you connect the PiKVM video? To VGA or DisplayPort output?

alexisfrjp, 2025-01-04:
I use the displayport output with the same hdmi/DP adapter I use for my external monitor.
Please let me know if I can provide any logs or anything.

mdevaev, 2025-01-04:
I'm starting to think that maybe your adapter has its own ideas about EDID. Or the BIOS wants to give 1080p 60Hz ignoring EDID. If the second is true, then there's nothing to be done about it, because it's a buggy BIOS :/

In the first case, changing the DisplayPort adapter to some other one may help. Also please try the default V3 EDID with it: `kvmd-edidconf --import-preset=v3`, and reboot PiKVM.

mdevaev, 2025-01-24:
Sup?

closed 2025-02-03 by mdevaev