pikvm/pikvm #1442: V3 image without the iso partition

theking8713, 2024-12-17:
Id like the ability to either generate an image without the iso partition and just the fat and root partititon for devices that utilize the cm4, or the availability of an lite image that doesn't have this massive bloated partiton that forces us to use cm4s with more than 8gb of emmc, since the kvmd can easily run on a 2gb ram module it only makes sense because most of these decides that use the cm4 also instantiate a USB hub pass threw like the x680 for example and with that it makes more sense to use USB media via the hub, like a USB SSD with tools like ventoy, this is my situation, I have a couple 2gb cm4s with only 8gb of emmc because there cost effective when considering the cost of the pikvm hardware the total solution is a more palatable price...  But the only images I can find extract to more than 8gb in size so I'm stuck either buying an overpriced model with more ram than what's needed or the rare 2gb models with 16 or 32gb of emmc only to have most of that space not needed on this immutable install... 
See what I'm getting at?
The regular v3 assembled image would be perfect minus the last partition that I end up removing anyway,
Iv been diligenty trying to image my 32gb model after removing this last partition so I can generate an img file from that and install that onto either 8gb model but I'm making zero progress

theking8713, 2024-12-18:
O and yes Iv attempted to build my own image several and it seems that there are a lot to f dependency related errors and then the make is and also make image step fails as the containers fails to locate the necessary filesystems to image ... Idk what is broken with that but it seems to me that the whole process is obfuscated within a docker container and the only thing I can find thus far that seems like it may result in an image that is "missing" the MSD partition is to enter the output files and edit the v3.config in disks and remove the step that creates the last partition for MSD, and no this process failed for me on both the fresh wsl instance and also a fresh install of rasbian cli 64bit on my 32gb cm4...
I'm at a total loss here and now iv got 3 cm4s and the geekworm x680 and a blikvm pcie version that I think is the x650 v1 from AliExpress, the version that isn't pulling power from a pcie slot but gets it's power from external type C... All sitting in a box not being deployed because Iv hit a wall with this step that seems like it should be trivial but is infact over my head...
I realize that this MSD feature is valued by some users but for me it seems like a buggy feature from all the forums iv seen especially on this particular geekworm hardware and it just makes infinatly more sense to me to use the USB host feature of most of these devices to pass threw a real hardware based block device that can install any os that ventoy can, including OEM iso of windows, any Linux distro, and any bsd base making it a far better solution than this kind of virtualized but not whole block device part of another systems block device were you may have more than one system sending I/O commands to the same block device because one partition is isolated and served as a otg when it works and you sacrifice the right mule to the exact right constellation that morning....no it's not worth the trouble and it's even less worth the trouble for users like me who have hardware that exposes a USB hub to the kvm that works well...
I don't know any thing here but if we "needed" a virtual CD rom mounted on a managed device couldnt that just be instantiated via a one time inflated image in cowspase it zram... That seems like a better solution than writing it to already questionable stability SD nand and it's very very low write cycle lifespan... I'm just sayin... 
But for me just a basic image that I can use raspberry pi imager to flash the emmc of a 2gb ram 8gb emmc RPI cm4 would be a godsend, 
As for now I have a couple little sipeed nanokvms and a jetkvm that seems to be just enough hardware and firmware to be a cost effective alternative to expensive ipmi enabled server hardware and I also have a cheap pikvm a4 deployed on a standalone appliance that is acting as a network and storage solution
But I want to be able to use this very expensive x680 and x650 hardware I have instead of the unstable nanokvms but unless something changes, with the size and bloat of the V3 images that just not going to happen, I can't see wasting a 8gb ram 32gb emmc model on effectively a transit device like a kvm and not using it as an actual compute target for services or as a workstation...
ANY help with this would be tremendously appreciated...
Like maybe an image I can test...
Or at least a brief educational on what exactly I'm doing wrong when trying to build the image myself...

mdevaev, 2024-12-18:
Hello.

V3 and V4 images are designed to run on our barnded PiKVM hardware. None of the official DIY builds involve using CM4 either. What you are asking for is required for third-party hardware that is not supported. Accordingly, we do not fulfill such requests. If you are experiencing problems with third-party hardware, contact its manufacturers.

closed 2024-12-18 by mdevaev

theking8713, 2024-12-20:
I realized that what I was asking was for a non official hardware but in reality I was just asking for generic images that are without the MSD features and partition... That should be interpreted as an assumed risk by the users ambiguous usecase not related to the actual hardware target at all beyond the fact that it's a broadcom arm core...
That being sead Iv solved my own issue with a brute force route by simply removing that part of the image and flashing as is with no verification and then going into the root filesystem and modifying the fstab to reflect the missing partition and also disabling the MSD feature in the requisite yaml... And then reinflating the filesystem to the remainder on the emmc(was only about 800MB short of the entire device after removing the MSD partition) so this ends up being a perfect solution for users that are using "non official" hardware but on platforms that require a cm4 and they can now use the commodity lowest spec cm4 that is arguably the most common ram and emmc size even now several years after the shortage...
I mean no animosity to anybody working on this project officially and I do infact only want to help wherever I can and this I feel is helpful so ya... Also thanks to this community for all the hardwork, these open "BMC" options are an absolute godsend vs extortionatly expensive traditional embedded BMC enabled server options that has and will continue to help everyday folks leverage affordable off the shelf hardware in ways that would have cost either a mortgage or a kidney to have access to before, be that for educational or entreprenial endeavors the value in this cannot be overstated ... Much Apreciated