pikvm/pikvm #1450: hide the remote cursor

cppcoffee, 2024-12-25:
**Describe the bug**
In the process of using PiKVM, there is an issue with mouse lag. The actual mouse has already reached the specified position, but the video transmission delay causes this situation.

When creating the HID-Mouse, is there a way to hide the remote mouse? If only the control-side web displays the mouse, it could reduce the mouse-related video transmission and eliminate the perception of lag.

**To Reproduce**
Moving the mouse shows a noticeable delay compared to the local mouse, even though the actual mouse has already reached the controlled PC's position.

**Expected behavior**
Hide the remote mouse and display the 'small blue dot' as the current mouse.

**Desktop:**
 - OS: Windows 11
 - Browser chrome 131.0.6778.205

**PiKVM info:**

```
# pacman -Q | grep kvmd
kvmd 3.333-1
kvmd-fan 0.30-1
kvmd-oled 0.27-1
kvmd-platform-v2-hdmiusb-rpi4 3.333-1
kvmd-webterm 0.50-1

# pacman -Q | grep ustreamer
ustreamer 6.11-1

# uname -a
Linux pikvm 6.6.27-current-sunxi64 #1 SMP Sat Apr 13 19:07:41 CST 2024 aarch64 GNU/Linux
```

cppcoffee, 2024-12-25:
Can the mouse be modified into a tablet mouse?

mdevaev, 2024-12-25:
PiKVM can't change cursor on remote machine because the link with it is just USB HID. There are two points:
1. When performing a click, the click will always occur in the place where the local blue dot points, even if the remote cursor has not yet reached there. When clicked, the current coordinates are immediately transmitted, followed by the click itself. Knowing this, you just need to get used to it a little and your work will be fast.
2. If you want to focus on only one cursor, you can disable the local blue dot.

lodibrahim, 2024-12-26:
You can hide remote cursor by downloading and setting invisible cursor

cppcoffee, 2024-12-26:
Hi @lodibrahim , Can PiKVM automatically hide remote cursor by downloading and setting invisible cursor?

Or do we need to manually operate the remote PC operating system to set the mouse to be hidden?

mdevaev, 2024-12-26:
No, PiKVM can't download and change os settings automatically **because the link with it is just USB HID**.

You need to perform it manually. Or you can use any of described workflow.

cppcoffee, 2024-12-26:
If we don't use the web and instead use VNC as the client, would it be possible to hide the mouse on the remote PC?

mdevaev, 2024-12-26:
No.

closed 2024-12-26 by mdevaev