pikvm/pikvm #1451: Japanese paste function clicks paste, but nothing happens.

InnerCalmer, 2024-12-26:


mdevaev, 2024-12-26:
[image]

mdevaev, 2025-01-02:
I checked and it can't be fixed due to absense of jp keymaps. At least now.

NiKiZe, 2025-01-02:
You would also need to know which IME method is enabled on the host, Kanji selection etc will also be difficult, if not impossible, (or maybe with some kind of OCR feedback to the input system)

Often Japanese is input by typing western chars, and you then get a menu to select which character to use, before moving to the next one (or set of characters)
There is modes for direct hiragana input, but knowing which method/mode is currently selected could be something the user needs to sync. It is not uncommon for the input system to be adaptive, and change priority of items in the selection list based on previous selections.

closed 2025-10-26 by mdevaev