pikvm/pikvm #1452: Reverse proxy works and page loads, but PiKVM doesn't connect

chriexpe, 2024-12-29:
This is a weird issue, it used to work just fine on this setup with reverse proxy using NGINX-PM (+ cloudflare for DNS).
The interface does load, I can login and access console (that just display "Reconnect") and KVM that only shows it's connecting (that orange Ethernet icon).

[image]

[image]

NGINX-PM Settings:
[image]
[image]

 - OS: All devices I tried, Android and Windows
 - Browser: Firefox, Edge, also both on mobile data

**PiKVM info:**
  - Raspberry Pi 4b 8Gb
  - PiKVM platform: v3 Hat
  - KVMD version: `kvmd 4.32-1 | kvmd-fan 0.31-1 | kvmd-platform-v3-hdmi-rpi4 4.32-1 | kvmd-webterm 0.50-1`
  - uStreamer version: `6.21-1`
  - Linux kernel: `6.6.45-8-rpi #1 SMP Wed Oct  2 19:56:38 UTC 2024 armv7l GNU/Linux`
  
I do have other things that are reverse proxied and they all work fine, even a WebRTC NVR.
Yes using local IP it works perfectly.

mdevaev, 2024-12-30:
Hello. Some URLs require websockets passing, so you need to enable it on your proxy.

chriexpe, 2024-12-30:
> Check JS browser console for errors.

[image]

mdevaev, 2024-12-30:
It looks like this is a proxy problem. It's breaking websocket, as I thought.

chriexpe, 2025-01-05:
> It looks like this is a proxy problem. It's breaking websocket, as I thought.

Indeed, later I found out that Authentik's custom configuration for NGINX was missing some header, after adding it everything is working again, thanks!

For anyone that happens to stuble on this, just add `proxy_set_header Upgrade $http_upgrade;` below "location / {"
Like this:
```
location / {
    # Put your proxy_pass to your application here
    proxy_pass          $forward_scheme://$server:$port;
    # Set any other headers your application might need
    # proxy_set_header Host $host;
    # proxy_set_header ...
```

closed 2025-01-05 by chriexpe

ljo123, 2025-01-13:
> > It looks like this is a proxy problem. It's breaking websocket, as I thought.
> 
> Indeed, later I found out that Authentik's custom configuration for NGINX was missing some header, after adding it everything is working again, thanks!
> 
> For anyone that happens to stuble on this, just add `proxy_set_header Upgrade $http_upgrade;` below "location / {" Like this:
> 
> ```
> location / {
>     # Put your proxy_pass to your application here
>     proxy_pass          $forward_scheme://$server:$port;
>     # Set any other headers your application might need
>     proxy_set_header Upgrade $http_upgrade;
>     # proxy_set_header Host $host;
>     # proxy_set_header ...
> ```

Thanks! I just updated PiKVM and got this error. Everything had been working fine previously. Can you share a bit more detail on your NPM custom location config?

ljo123, 2025-01-13:
Thanks but my issue is something different. Everything was working but then i ran the pikvm update function and now I have the error. I can connect directly to the local IP and it works but through NPM it won't connect to the remote device.