pikvm/pikvm #1455: After plugging in a 1920x1080 monitor and then plugging back in the 1600x1200 piKVM then displays the 1080p image across the 1600x1200 screen window

meyerkev, 2025-01-08:
**Describe the bug**
My KVM appears to be smearing 1080p all over a 1600x1200 kvm window

**To Reproduce**
Steps to reproduce the behavior, like:
1. Have a server, in this case a Dell R640, but it works with any server
2. Plug in a 1080p monitor for any reason
3. Plug the KVM back into that server
4. The server now looks like that screenshot where a 1080p video stream is being collapsed down into a 1080p window

**Expected behavior**
I either need a way to make the window 1080p so it doesn't smear or somehow convince this USB drive I'm using to install Ubuntu (and many many other contexts) that it wants to output 1600x1200

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 LTS, but it happens everywhere
 - Browser: Chrome
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version:  Raspberry Pi Compute Module 4 Rev 1.1
  - PiKVM platform [e.g. v2-hdmi]: v4mini-hdmi
  - Video capture type: VGA to HDMI to Teksmart switch to this
  - KVMD version: 
```
kvmd 4.38-1
kvmd-fan 0.31-1
kvmd-platform-v4mini-hdmi-rpi4 4.38-1
kvmd-webterm 0.50-1
```
  - uStreamer version: `ustreamer 6.21-2`
  - Linux kernel: `Linux pikvm 6.6.45-9-rpi #1 SMP Mon Dec 30 22:25:40 UTC 2024 armv7l GNU/Linux`

**Additional context**
Add any other context about the problem here.

mdevaev, 2025-01-08:
As far as I can see, you are using a VGA converter, which adds some mess to the situation. The server was configured for one video mode, then you replaced its output to a VGA-HDMI device, and it no longer works.

The signal resolution selects the source, all PiKVM can do is display it. If you see such a picture, it means that the source is lying about the resolution used, this usually happens with bad converters. I recommend simply restarting the server so that the entire video chain can re-negotiate the resolution used, perhaps this will fix the situation.

meyerkev, 2025-01-08:
That is at least half the fun.  The resolution issues persist across cold boots.  

But ok.  Bad VGA to HDMI adapter.  That's worth checking.

mdevaev, 2025-01-11:
Okay. Please join our [discord](https://discord.com/invite/bpmXfz5) for further support, github is only for bugs and developing.

closed 2025-01-11 by mdevaev