pikvm/pikvm #1456: Home set up

scochavi, 2025-01-08:
I have a IPKVM V4 mini it’s hooked up to MacBook login for the system works all good correct the wires are connected, but the keyboard amount functionality will not respond

https://github.com/user-attachments/assets/508e5fcc-86fb-4b75-9e12-49ec302039da

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
  - uStreamer version: `pacman -Q | grep ustreamer`
  - Linux kernel: `uname -a`

**Additional context**
Add any other context about the problem here.

scochavi, 2025-01-08:
Yes, I’m using a USB type C to USB type C cord I saw on other forms that it could be the wire so I ordered other USB type C wires hopefully that fixes or resolve my issue if not, could be any settings on the Mac because if you saw the video, I posted earlier when I plug it in and recognizes the cable but then quickly The pop-up that’s mentioning about the accessories quickly disappeared, so it doesn’t allow me to set it up

arch1mede, 2025-01-08:
Yes, we have also seen this behavior, have you tried what's linked for the Source within that same FAQ entry? You are dealing with a MAC OSX behavior.

For some this appears and allows you to continue, for others it briefly pops up and goes away and you have to manually go in and make some changes. I would try this first before ordering this cable. You might have a data capable cable already.

mdevaev, 2025-02-03:
@scochavi Sup?

mdevaev, 2025-02-09:
Closing due no feedback. Please feel free to contact us again.

closed 2025-02-09 by mdevaev