pikvm/pikvm #1461: unable to satisfy dependency 'python<3.13' required by kvmd

rmbleeker, 2025-01-21:
**Describe the bug**
When trying to do an update of my PiKVM (v3) I was hit with a dependency conflict. It appears the latest version of kvmd requires a Python version that conflicts with the version required by several other packages.

**To Reproduce**
1. Log in to your pikvm
2. Enable read-write mode (`rw`)
3. Perform an update using either `pacman -Syu` or the `pikvm-update` command
4. Notice the dependency errors such as `installing python (3.13.1-1) breaks dependency 'python<3.13' required by ...`

**Expected behavior**
For the pikvm to be able to update it's packages, especially those belong to PiKVM themselves.

**Screenshots**
[image]

**PiKVM info:**
  - Raspberry Pi 4
  - PiKVM v3-hdmi
  - Video capture CSI bridge
  - KVMD version: 4.42-1
  - uStreamer version: 6.22-1
  - Linux kernel: `Linux <hostname> 6.6.45-9-rpi #1 SMP Mon Dec 30 22:25:40 UTC 2024 armv7l GNU/Linux`

**Terminal output:**
```
[root@pikvm ~]# rw
+ mount -o remount,rw /
+ mount -o remount,rw /boot
+ set +x
=== PiKVM is in Read-Write mode ===
[root@pikvm ~]# pacman -Syu
:: Synchronizing package databases...
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
 aur is up to date
 pikvm is up to date
:: Starting full system upgrade...
resolving dependencies...
looking for conflicting packages...
error: failed to prepare transaction (could not satisfy dependencies)
:: unable to satisfy dependency 'python<3.13' required by kvmd
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-luma-core
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-luma-oled
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-pyghmi
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-pyrad
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-pyserial-asyncio
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-raspberry-gpio
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-smbus2
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-spidev
[root@pikvm ~]# pikvm-update 
+ trap on_error ERR
+ _yes='--noconfirm --ask=4 --overwrite \*'
+ rw
+ mount -o remount,rw /
+ mount -o remount,rw /boot
+ set +x
=== PiKVM is in Read-Write mode ===
+ pacman -Syy
:: Synchronizing package databases...
 core                                            238.1 KiB   301 KiB/s 00:01 [###########################################] 100%
 extra                                             9.2 MiB  12.1 MiB/s 00:01 [###########################################] 100%
 community                                        45.0   B   214   B/s 00:00 [###########################################] 100%
 alarm                                            95.7 KiB   456 KiB/s 00:00 [###########################################] 100%
 aur                                               9.3 KiB  44.5 KiB/s 00:00 [###########################################] 100%
 pikvm                                            11.4 KiB  71.2 KiB/s 00:00 [###########################################] 100%
++ pacman -S -u --print-format %n
++ grep -v pikvm-os-updater
++ wc -l
error: failed to prepare transaction (could not satisfy dependencies)
+ '[' 9 -eq 0 ']'
+ rm -rf /var/cache/pacman/pkg
+ mkdir -p /var/cache/pacman/pkg
+ '[' -z '' ']'
++ pacman -S --needed --print-format %n pikvm-os-updater
++ wc -l
+ '[' 0 -ne 0 ']'
+ grep -q '^C.UTF-8 UTF-8' /etc/locale.gen
+ for _pkg in rpi-eeprom rpi4-eeprom
+ pacman -Q rpi-eeprom
+ for _pkg in rpi-eeprom rpi4-eeprom
+ pacman -Q rpi4-eeprom
+ pacman -Qi python-ajsonrpc
+ grep Depends
+ grep -q -e 'python<3\.11' -e 'python<3\.12'
error: package 'python-ajsonrpc' was not found
+ pacman -Q raspberrypi-utils libgpiod
+ pacman --noconfirm --ask=4 --overwrite '\*' -Su
:: Starting full system upgrade...
resolving dependencies...
looking for conflicting packages...
error: failed to prepare transaction (could not satisfy dependencies)
:: unable to satisfy dependency 'python<3.13' required by kvmd
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-luma-core
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-luma-oled
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-pyghmi
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-pyrad
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-pyserial-asyncio
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-raspberry-gpio
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-smbus2
:: installing python (3.13.1-1) breaks dependency 'python<3.13' required by python-spidev
++ on_error
++ set +x
==============================================================
      An unexpected error occurred during the update.
==============================================================

==============================================================
Please note that the filesystem now remain in Read-Write mode.
       A reboot is necessary to make it Read-Only again.
The reboot can be performed later using the 'reboot' command.
==============================================================
++ set -x
[root@pikvm ~]# 
```

mdevaev, 2025-01-21:
Hello. Upstream updated python version. Please wait, I'll update our packages too.

mdevaev, 2025-01-21:
Fixed, just perform `pikvm-update`.

closed 2025-01-21 by mdevaev

rmbleeker, 2025-01-21:
Confirmed, updates work again. Thanks for the quick fix.

mdevaev, 2025-01-22:
👌