pikvm/pikvm #1462: BaseURL configuration for Reverse Proxy with NGINX or others

MichaelWoodc, 2025-01-22:
When I setup my network, I would like to have my pikvm accessible with a different base url or path so that I can have only one or two ports open to the outside world.  In other services like Navidrome I can easily set up a base path, like this:

[image]

**Describe the solution you'd like**
I'd like to have a super easy place to setup a path for a reverse proxy like this

**Describe alternatives you've considered**
I've tried complicated reverse proxy setup, like this, but it didn't work.  I got so close, but the post address for the login was incorrect and a few other things:
```
# PiKVM Proxy Configuration
location /pi_kvm/ {
    # Forward to PiKVM over HTTPS
    proxy_pass https://192.168.1.127/;  # Forward to PiKVM

    # Disable SSL verification for self-signed certificates
    proxy_ssl_verify off;

    #  (not sure this is right)
    proxy_set_header Host $host;  # Pass the original host
    proxy_set_header X-Real-IP $remote_addr;  # Preserve the client IP
    proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;  # Add any proxy IPs
    proxy_set_header X-Forwarded-Proto $scheme;  # Pass the protocol

    # WebSocket support
    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection "upgrade";
    proxy_cache_bypass $http_upgrade;

    # Rewriting paths so PiKVM receives the right URLs and resources
    rewrite ^/pi_kvm(/.*)$ $1 break;

    # Ensure any redirects from PiKVM are handled correctly
    proxy_redirect / /pi_kvm/;

    # Handle the PiKVM static resources like CSS, JS, and images
    sub_filter_once off;  # Allow multiple replacements in a response
    sub_filter '/share/' '/pi_kvm/share/';  # Replace /share/ with /pi_kvm/share/
}
```

**Additional context**
The only thing I've been able to do is do a subdomain.  This works great.  I'd prefer to be able to implement it this other way, however.  

[image]

MichaelWoodc, 2025-01-23:
I did some exploring today, it seems there might be an easy way.  I went to the web terminal and did:
`find` / -name "*nginx*" 2>/dev/null

That returns everything having to do with nginx.  I popped the most likely file into the editor
```
su - (take root)
rw (make fs rear write)
nano /etc/nginx/nginx.conf
```

I'm not entirely faililiar with nginx yet, but it seems this might be a place to edit the base URL.  Some experimenting is necessary.

[nginx.conf.txt](https://github.com/user-attachments/files/18526026/nginx.conf.txt)

MichaelWoodc, 2025-01-23:
A little more digging and I have found a lot of files in: /etc/kvmd/nginx
that seem to have everything to do with it.  

[image]

It looks like I could just add /pi_kvm everywhere there.  Is that correct?

mdevaev, 2025-01-27:
Don't do this, you can misconfigure something and pikvm will break. We will make specific fix for this problem.

MichaelWoodc, 2025-02-01:
> Don't do this, you can misconfigure something and pikvm will break. We will make specific fix for this problem.

Please let me know if and how I can help out on this one!  And thanks guys!

mdevaev, 2025-02-01:
I'm working on it now.

mdevaev, 2025-02-02:
Okay, it's done. Please update OS. No additional settings are required for base url.

I also wrote a small guide for reverse proxying with two tested example configs: https://docs.pikvm.org/reverse_proxy/

MichaelWoodc, 2025-02-04:
Thanks! I hit that button on the main page, hopefully it's enough to get a coffee where you're at.  Saved me a lot of work!

mdevaev, 2025-02-04:
👌

fallenleavesgocrunch, 2025-02-10:
This hasn't worked for me. When I go to /pikvm/ it redirects back to /login with the handle_path directive in caddy.

mdevaev, 2025-02-10:
Yes, it seems I missed it with Caddy. PiKVM has a login redirect location in the internal nginx:
```nginx
location @login {
    return 302 /login;
}
```

To handle it in the proxy Nginx I used this:
```nginx
proxy_redirect ~^(/.*)$ /pikvm$1;
```

We need something similar on Caddy. Do you know about it? I don't have much Caddy experience.

In the meantime, I will try to find a way to change this redirect by making it relative on the PiKVM side.

mdevaev, 2025-02-10:
@fallenleavesgocrunch I found a solution. You need to add `header_down` instruction:

```nginx
https://mydomain {
handle_path /pikvm/* {
    reverse_proxy https://pikvm.local {
        transport http {
            tls_insecure_skip_verify  # Same behaviour as Nginx
        }
        header_up Host {upstream_hostport}
        header_down Location "^(/.*)$" "/pikvm$1"
    }
}
```

I've tested it and updated documentation.

closed 2025-02-10 by mdevaev

fallenleavesgocrunch, 2025-02-10:
Wow that was fast! Thank you :)

mdevaev, 2025-02-10:
Ur welcome 👌