pikvm/pikvm #1463: Keystrokes sent repeatedly all the time

Simon1511, 2025-01-24:
**Describe the bug**
Pressing any key on the keyboard will make it "spam" that key for around 1-2 seconds. So for example, when I try to log in as "root" in a shell, I'll end up with "rrrrrrrrrrrrrrrrrrrooooooottttt". I already tried enabling the recently added "Bad link mode" in the PiKVM UI, but that does not fix the issue at all. In fact it makes it worse because both the physical and virtual keyboard will eventually just stop accepting any key input until I switch the inputs of my KVM-Switch (ezcoo EZ-SW41H21-KVMU3P) and then switch back (I guess this is equivalent of unplugging USB from the physical machine?).

**To Reproduce**
1. Plug a PC into the PiKVM (HDMI + USB-A to USB-A cable)
2. Open the PiKVM Web UI
3. Try to type anything and it'll "spam" that key with no way of interrupting it.

**Expected behavior**
Any text should be written as normal, it should only output what I'm actually typing instead of repeating it.

**Screenshots**
//

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Firefox
 - Version: 134
 - VNC client: none

**PiKVM info:**
  - Raspberry Pi board version: RPI 4b
  - PiKVM platform: /
  - Video capture type: CSI bridge
  - KVMD version: 4.43-1
  - uStreamer version: 6.23-1
  - Linux kernel: Linux pikvm 6.6.45-9-rpi

**Additional context**
I'm using the ezcoo EZ-SW41H21-KVMU3P KVM-Switch with my PiKVM. Not sure if that's related to the issue.

Simon1511, 2025-01-26:
Seems like none of my devices like PiKVM's mouse? Once I added keybord, mouse and MSD to override.yml and toggled off mouse in the Web UI it started to work perfectly fine, no more repeated key presses.

Perhaps someone knows why specifically the mouse triggers this?

mdevaev, 2025-01-26:
What the host device do you use?

Simon1511, 2025-01-26:
> What the host device do you use?

Multiple different hosts, happens with all of them:
- Fujitsu Q556
- Fujitsu S920
- Dell OptiPlex 7040M
- some noname chinese Mini-PC

mdevaev, 2025-01-27:
Please check the following:
* Will this bug be reproduced without ezcoo, just with direct pikvm->host connection?
* Please try other browser, like chrome without any extensions and even in incognito mode.

mdevaev, 2025-01-31:
Sup?

Simon1511, 2025-02-01:
> Please check the following:
> 
>     * Will this bug be reproduced without ezcoo, just with direct pikvm->host connection?
> 
>     * Please try other browser, like chrome without any extensions and even in incognito mode.

Sorry for the long delay on this, been pretty busy with other stuff the past week.

I just tried what you suggested, and in fact that issue does not happen at all with direct PiKVM to Host connection, regardless of the browser used.

With the Ezcoo connected I tried Firefox and Edge, both have the issue.

On a side note: with the PiKVM's USB plugged into the MOUSE port of the Ezcoo, the keyboard will function correctly, however the mouse will stop working and switching between inputs is not possible.
With it plugged into the KEYBOARD port on the Ezcoo, all works but said issue happens (and goes away once I turn off the mouse in PiKVM).

mdevaev, 2025-02-02:
Okay, this is Ezcoo problem, it can't be solved on our side. Apparently, their USB driver inside the device does not handle mouse events correctly.

I would recommend that you use a cable and configure port switching from PiKVM: http://docs.pikvm.org/ezcoo

Simon1511, 2025-02-02:
> Okay, this is Ezcoo problem, it can't be solved on our side. Apparently, their USB driver inside the device does not handle mouse events correctly.
> 
> I would recommend that you use a cable and configure port switching from PiKVM: http://docs.pikvm.org/ezcoo

This specific Ezcoo switch does not have a dedicated control port though. I got it for cheap because it's listed as "Everything is working as expected" on the [PiKVM Docs](https://docs.pikvm.org/multiport/#list-of-tested-third-party-kvms):

[image]

 Apparently that's not the case. You might want to add something like "Mouse input broken" to the Docs then.

On a side note though: Would it be possible to split the PiKVM's USB output into two cables, one for keyboard + MST and one for mouse? Maybe even with an Arduino or something in between?

I also wonder if it would help to modify the HID devices that PiKVM outputs? Currently it seems to be detected as "Microsoft-KM-v1.a".

mdevaev, 2025-02-02:
> Apparently that's not the case. You might want to add something like "Need to disable Mouse input in PiKVM" to the Docs then.

We have not been informed of any problems before. Maybe Ezcoo has updated the firmware, or vice versa, your firmware is too old.

> On a side note though: Would it be possible to split the PiKVM's USB output into two cables, one for keyboard + MSD and one for mouse?

No. But I was thinking, what happens if you disable MSD? Would it work better with just a keyboard and mouse?

> I also wonder if it would help to modify the HID devices that PiKVM outputs?

It will not help. PiKVM HID is already configured for maximum compatibility. You can only try disabling horizontal scrolling and double mouse: https://docs.pikvm.org/mouse

Simon1511, 2025-02-02:
> No. But I was thinking, what happens if you disable MSD? Would it work better with just a keyboard and mouse?

Disabling MSD has no effect, it will continue to repeat the keys even with it disabled.

> It will not help. PiKVM HID is already configured for maximum compatibility. You can only try disabling horizontal scrolling and double mouse: https://docs.pikvm.org/mouse

I already tried the dual mode thing but unfortunately having both enabled does not help, I need to have both relative and absolute mouse disabled to fix that issue:

[image]

This is my override.yaml, not sure if this is fully correct:

```
kvmd:
    hid:
        mouse:
            absolute: false
            horizontal_wheel: false
        mouse_alt:
            device: /dev/kvmd-hid-mouse-alt
    gpio:
        drivers:
            otgconf:
                type: otgconf
        scheme:
            hid.usb0:
                driver: otgconf
                mode: output
                pin: hid.usb0
                pulse: false
            hid.usb1:
                driver: otgconf
                mode: output
                pin: hid.usb1
                pulse: false
            hid.usb2:
                driver: otgconf
                mode: output
                pin: hid.usb2
                pulse: false
            mass_storage.usb0:
                driver: otgconf
                mode: output
                pin: mass_storage.usb0
                pulse: false
        view:
            table:
                - ['#Keyboard', '#hid.usb0', hid.usb0]
                - ['#Absolute Mouse', '#hid.usb1', hid.usb1]
                - ['#Relative Mouse', '#hid.usb2', hid.usb2]
                - ['#Mass Storage Drive', '#mass_storage.usb0', mass_storage.usb0]
```

mdevaev, 2025-02-02:
> Do you possibly know if it's possible to check what firmware version mine is running or something? Or do you know others with this specific KVM-Switch that I could ask for help?

No, I don't know. @arch1mede do you know someone? Or maybe do you have this switch?

To be honest, if I were you, I would return it and replace it for an Ezcoo with a control port.

Simon1511, 2025-02-02:
> To be honest, if I were you, I would return it and replace it for an Ezcoo with a control port.

Unfortunately bought it used from Kleinanzeigen (German counterpart of Facebook Marketplace) for 90€. I don't really need the mouse that often as my hosts are all terminal only, so it's not much of a big deal. I'll keep my eyes open for some good offer on another Ezcoo one, or even the official PiKVM switch ;)

Thanks a lot for your help though, I'll see if I can play around with it a little more and will eventually get back here if I ever figure something out.

mdevaev, 2025-02-02:
I'm sorry that I couldn't solve your problem. I remembered that we have an old patch that allowed us to use otg and arduino (now the pico) hid at the same time. But it doesn't accept the upstream and it takes a lot to adapt it to make it work. So yes, another switch would be the easiest solution.

Simon1511, 2025-02-02:
> By the way, can you solder? If so, I have a tricky but working option for you.

Don't have a soldering iron here at the moment and my soldering skills aren't great but I might still try whatever your suggestion is ;)

mdevaev, 2025-02-02:
In short, you can take several optocouplers and resistors and solder them to the Ezcoo board so that they close the switching buttons. Then connect the optocoupler with a resistor to the GPIO. The circuit is identical to ATX, just select unused pins.

Simon1511, 2025-02-02:
Yeah that sounds way too advanced for me 😂

Thanks for the suggestion though.

mdevaev, 2025-02-02:
At least I tried :D

Okay, then I'm closing this issue. If you have any new info, you can let me know and I will reopen it if necessary.

closed 2025-02-02 by mdevaev

Simon1511, 2025-11-25:
> [@Simon1511](https://github.com/Simon1511) Sorry for reopening the topic but I would like to know which features dont work with the ezcoo [EZ-SW41H21-KVMU3P 8K 4x1 switch](https://www.amazon.de/Computer-Tastatur-HDCP2-2-Fernbedienung-Kabelx4/dp/B0F4R9X7VG)? It's one of the few compatible HDMI 2.1 KVMs which are relative affordable in Germany. Is it just mouse support? Does switching the HDMI inputs still work?

Switching the HDMI inputs works fine, though sometimes after switching it doesn't pick up the input correctly. But that can be fixed by hitting CTRL + CTRL + ESC once after switching to "reset" the HID on the KVM.

Mouse support however is completely borked with that KVM for me. As said earlier in this thread it might just be an issue with some models having a newer firmware or something.

However: Keyboard input will be completely broken unless you disable mouse input completely using the GPIO menu. If you keep the mouse enabled any keyboard input will end up being "delayed" and repeated (for example: press the M key -> nothing happens -> press again -> wait -> it repeadetly prints out "M" unless you press another key -> whole thing repeats from the start).
The GPIO menu can be enabled by adding the following to your /etc/kvmd/override.yml:

```
kvmd:
    hid:
        mouse:
            absolute: false
            horizontal_wheel: false
        mouse_alt:
            device: /dev/kvmd-hid-mouse-alt
    gpio:
        drivers:
            otgconf:
                type: otgconf
        scheme:
            hid.usb0:
                driver: otgconf
                mode: output
                pin: hid.usb0
                pulse: false
            hid.usb1:
                driver: otgconf
                mode: output
                pin: hid.usb1
                pulse: false
            hid.usb2:
                driver: otgconf
                mode: output
                pin: hid.usb2
                pulse: false
            mass_storage.usb0:
                driver: otgconf
                mode: output
                pin: mass_storage.usb0
                pulse: false
        view:
            table:
                - ['#Keyboard', '#hid.usb0', hid.usb0]
                - ['#Absolute Mouse', '#hid.usb1', hid.usb1]
                - ['#Relative Mouse', '#hid.usb2', hid.usb2]
                - ['#Mass Storage Drive', '#mass_storage.usb0', mass_storage.usb0]
```

That will give you the opportunity to disable Absolute and Relative mouse via the GUI:

[image]

After doing that the keyboard will work as normal, but mouse will be completely disabled of course. Note that, after rebooting the PiKVM, it automatically switches both settings back to "On".

mdevaev, 2026-04-18:
May I ask you for PR? I think the reason is firmware.