pikvm/pikvm #1470: Support for IPCameras

EvanCarroll, 2025-02-18:
I have multiple cheap Chinese IP Cameras. These are all easily detected and functioning with something like TinyCam. Technically an IPCamera just outputs mjpeg or h.264. 

It would be very cool if I could set the h.264 stream in pikvm and export that camera as a USB camera to the device. Then multiple pikvm devices could share the same h.264 stream.

mdevaev, 2025-03-09:
What is usecase of this?

EvanCarroll, 2025-03-09:
Sharing one web camera with multiple pikvms so on any of them I can open up Teams/Slack and they can see my face without having to connect 1 usb camera to each pikvm.

mdevaev, 2025-03-09:
I don't understand your setup, do you keep several pikvms nearby?

mdevaev, 2025-03-15:
Sup

EvanCarroll, 2025-03-15:
@mdevaev I have multiple computers which share an ipcamera using v4l2loopback. I want to put a macbook behind a pikvm and share the same camera. There are other suggestions that would work too that like the idea of proxying the webcam from the browser to pikvm, but that's massively more complex. This solution would probably be far easier, even if it would only work with ip cameras.

For example. All You have to do to virutalize an ipcamera as a webcam on a Linux machine is,

    sudo modprobe v4l2loopback
    ffmpeg -rtsp_transport tcp -i "rtsp://192.168.1.100/stream" -f v4l2 /dev/video2

(assuming /dev/video2 is the virutal webcam you get). From this point the pikvm would have access to the ip camera, and presumably the rest of the plumbing is already there.

mdevaev, 2025-03-18:
Okay, I get it. Camera support may be added in the future, but it's a low priority right now.

Majestic7979, 2025-03-31:
This is actually a very cool idea! I would still like to forward the webcam on the computer I'm on to PiKVM via the browser because it's the only way I can work remotely and "check in" with employer without taking the corporate device with me, which is not allowed as it is geo-location locked. It's really important that I have webcam functionality... I hope it is prioritized more urgently just like remote microphone was a game changer.

FearNaBoinne, 2025-04-04:
I think it would be more useful to include USB passthrough, so you can forward a USB device to the remote host. That way you could forward any USB device, including a webcam, to any PiKVM. There is USB over Ethernet already, so it should be feasible. The hardest step would be the PiKVM to host step, I think?

EvanCarroll, 2025-04-04:
> I think it would be more useful to include USB passthrough, so you can forward a USB device to the remote host. That way you could forward any USB device, including a webcam, to any PiKVM. There is [...]

Has 0 to do with this issue. Forward a webcam from a computer with a browser isn't more useful then just allowing the pikvm access directly to the webcam's stream. And that requires _wayyy_ more work.

EvanCarroll, 2025-04-16:
I think this conversation has to get a lot more technical. To start off, there is two seperate things here, 

* "passthrough" creating a bridge with WebUSB
* A concept of sharing the webcam stream over websockets with something like [mediacapture-streams/](https://www.w3.org/TR/mediacapture-streams/)

Abstracted USB passthrough is a non-starter as I understand it. Any device registered as an HID device can not be seen by WebUSB. When you mention a solution like https://www.virtualhere.com/ you're speaking apples and oranges here. That solution does *NOT* run through the browser. In fact, they make a _different_ product for each different operating system which turns the USB Host into a server. When you point to this as an example of prior art, I'm confused at which I should read this,

* Naivety with regard to the technology
* Or as a serious suggestion
 
This kind of "solution" would require PiKVM literally

* writing tens of thousands of lines of code for all operating systems to share a USB Device
* to have every accessing device act as a USB server to share it's webcam
* Require the PiKVM device to act as USB client to that USB server
* And to resolve any networking issues related to connectivity

Imagine the UI too,

> Do you want to share your webcam with PiKVM, install this USB server software as root/admin that can communicate with your USB devices deep in the stack and that provides an external API to allow remote machines to communicate to them!

I think the whole thing is a non-starter.

----

When it comes to accessing a video device and proxying it over websockets or whatever, is possible but that's way more complex if for no other reason then because it would obviously require modification to the PiKVM webserver to handle an incoming video streams from websockets (including a recv and send buffer), reconnection, and all that other goodness and it would require working with browser finkeyness: for example, does this api work over http? Or just https (no idea but it wouldn't shock me if it required https to share a video stream). 

Compare that to merely redirecting the stream of the webcam to a loopback. That's dirt easy. Like I said, **two lines of code**

```sh
sudo modprobe v4l2loopback
ffmpeg -rtsp_transport tcp -i "rtsp://192.168.1.100/stream" -f v4l2 /dev/video2
```

That's really it. That creates a device on Linux that ffmpeg can write to, and ffmpeg already has the network plumbing to read from an rtsp stream. So long as your network ip cam provides an rtsp stream you're good! And better yet, nothing else in PiKVM has to be modified.

mdevaev, 2025-11-16:
I think there should be some kind of generic way to output connected cameras to the interface and control them. Perhaps it's just some kind of camera app from what's available in open source.

suakhtar, 2025-12-14:
How to access to one Pikvm ip camera as webcam , can anyone share steps to configure please ?