pikvm/pikvm #1471: Constant 'Unexpected logout occured, please log in again' messages, requiring me to log in again

PrimozR, 2025-02-20:
**Describe the bug**
Lately I have been getting often, but random logout events, requiring me to log back in.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to PiKVM KVM view, use the KVM, randomly get logged out.

**Expected behavior**
Normal use of KVM function without being logged out

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser: Brave
 - Version: 1.75.180

**PiKVM info:**
  - Raspberry Pi board version: RPi 4
  - PiKVM platform: V3 HAT
  - KVMD version: 4.58-1
  - uStreamer version: 6.31-1
  - Linux kernel: Linux pikvm 6.6.45-10-rpi #1 SMP Fri Jan 24 03:51:01 UTC 2025 armv7l GNU/Linux

mdevaev, 2025-02-20:
Please attach the output of `journalctl -u kvmd`

PrimozR, 2025-02-20:
Looks like I get logouts mostly on isntances where video output stops and restarts (reboots and during boots and the like).
I also sometimes get nginx errors (500) requiring me to restart pikvm...

`Jan 14 20:13:28 pikvm kernel: Booting Linux on physical CPU 0x0
Jan 14 20:13:28 pikvm kernel: Linux version 6.6.45-10-rpi (alarm@fb2050577948) (distcc 3.4 armv7l-unknown-linux-gnueabihf, GNU ld (GNU Binutils) 2.42.0) #1 SMP Fri Jan 24 03:51:01 UTC 2025
Jan 14 20:13:28 pikvm kernel: CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
Jan 14 20:13:28 pikvm kernel: CPU: div instructions available: patching division code
Jan 14 20:13:28 pikvm kernel: CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
Jan 14 20:13:28 pikvm kernel: OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
Jan 14 20:13:28 pikvm kernel: random: crng init done
Jan 14 20:13:28 pikvm kernel: Memory policy: Data cache writealloc
Jan 14 20:13:28 pikvm kernel: Reserved memory: bypass linux,cma node, using cmdline CMA params instead
Jan 14 20:13:28 pikvm kernel: OF: reserved mem: node linux,cma compatible matching fail
Jan 14 20:13:28 pikvm kernel: OF: reserved mem: 0x000000003ef64b00..0x000000003ef64eef (0 KiB) nomap non-reusable nvram@0
Jan 14 20:13:28 pikvm kernel: cma: Reserved 128 MiB at 0x0000000030000000 on node -1
Jan 14 20:13:28 pikvm kernel: Zone ranges:
Jan 14 20:13:28 pikvm kernel:   DMA      [mem 0x0000000000000000-0x000000002fffffff]
Jan 14 20:13:28 pikvm kernel:   Normal   empty
Jan 14 20:13:28 pikvm kernel:   HighMem  [mem 0x0000000030000000-0x0000000037ffffff]
Jan 14 20:13:28 pikvm kernel: Movable zone start for each node
Jan 14 20:13:28 pikvm kernel: Early memory node ranges
Jan 14 20:13:28 pikvm kernel:   node   0: [mem 0x0000000000000000-0x0000000037ffffff]
Jan 14 20:13:28 pikvm kernel: Initmem setup node 0 [mem 0x0000000000000000-0x0000000037ffffff]
Jan 14 20:13:28 pikvm kernel: percpu: Embedded 18 pages/cpu s41940 r8192 d23596 u73728
Jan 14 20:13:28 pikvm kernel: pcpu-alloc: s41940 r8192 d23596 u73728 alloc=18*4096
Jan 14 20:13:28 pikvm kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Jan 14 20:13:28 pikvm kernel: Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32>Jan 14 20:13:28 pikvm kernel: audit: disabled (until reboot)
Jan 14 20:13:28 pikvm kernel: Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
Jan 14 20:13:28 pikvm kernel: Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Jan 14 20:13:28 pikvm kernel: Built 1 zonelists, mobility grouping on.  Total pages: 227648
Jan 14 20:13:28 pikvm kernel: mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Jan 14 20:13:28 pikvm kernel: Memory: 741484K/917504K available (14336K kernel code, 1666K rwdata, 3968K rodata, 2048K init, 644K bss, 44948K reserved, 131072K cma-reserved, 0K highmem)
Jan 14 20:13:28 pikvm kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Jan 14 20:13:28 pikvm kernel: ftrace: allocating 41184 entries in 121 pages
Jan 14 20:13:28 pikvm kernel: ftrace: allocated 121 pages with 5 groups
Jan 14 20:13:28 pikvm kernel: trace event string verifier disabled
Jan 14 20:13:28 pikvm kernel: rcu: Hierarchical RCU implementation.
Jan 14 20:13:28 pikvm kernel:         Rude variant of Tasks RCU enabled.
Jan 14 20:13:28 pikvm kernel:         Tracing variant of Tasks RCU enabled.
Jan 14 20:13:28 pikvm kernel: rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Jan 14 20:13:28 pikvm kernel: NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jan 14 20:13:28 pikvm kernel: GIC: Using split EOI/Deactivate mode
Jan 14 20:13:28 pikvm kernel: rcu: srcu_init: Setting srcu_struct sizes based on contention.
Jan 14 20:13:28 pikvm kernel: arch_timer: cp15 timer(s) running at 54.00MHz (phys).
Jan 14 20:13:28 pikvm kernel: clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
Jan 14 20:13:28 pikvm kernel: sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
Jan 14 20:13:28 pikvm kernel: Switching to timer-based delay loop, resolution 18ns
Jan 14 20:13:28 pikvm kernel: Console: colour dummy device 80x30
Jan 14 20:13:28 pikvm kernel: printk: console [tty1] enabled
Jan 14 20:13:28 pikvm kernel: Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
Jan 14 20:13:28 pikvm kernel: CPU: Testing write buffer coherency: ok
Jan 14 20:13:28 pikvm kernel: CPU0: Spectre BHB: enabling loop workaround for all CPUs
Jan 14 20:13:28 pikvm kernel: pid_max: default: 32768 minimum: 301
Jan 14 20:13:28 pikvm kernel: LSM: initializing lsm=capability,integrity
Jan 14 20:13:28 pikvm kernel: Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
Jan 14 20:13:28 pikvm kernel: Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
Jan 14 20:13:28 pikvm kernel: cgroup: Disabling memory control group subsystem
Jan 14 20:13:28 pikvm kernel: CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Jan 14 20:13:28 pikvm kernel: RCU Tasks Rude: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
Jan 14 20:13:28 pikvm kernel: RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1.
Jan 14 20:13:28 pikvm kernel: Setting up static identity map for 0x200000 - 0x20003c
[... cut ...]
Feb 20 11:45:48 pikvm kvmd[516]: kvmd.plugins.msd.otg.storage      INFO --- => Remounting /var/lib/kvmd/msd to RW-mode ...
Feb 20 11:45:48 pikvm kvmd[516]: kvmd.plugins.msd.otg.storage      INFO --- => Storage in the RW-mode now
Feb 20 11:45:48 pikvm kvmd[516]: kvmd.plugins.msd.otg              INFO --- Writing 'gparted-live-1.5.0-6-amd64.iso' image (536870912 bytes) to MSD ...
Feb 20 11:46:03 pikvm sshd-session[769]: Accepted password for root from 10.0.0.11 port 53700 ssh2
Feb 20 11:46:03 pikvm sshd-session[769]: pam_unix(sshd:session): session opened for user root(uid=0) by root(uid=0)
lines 818-919/919 (END)`

mdevaev, 2025-02-20:
I don't see any problems here.

Please try to replace SD card and install new OS. Maybe it's just corrupted.

mdevaev, 2025-02-23:
Is it solved now?

PrimozR, 2025-02-24:
No. I reinstalled the OS and it is much better (no nginx errors) but I also got an unexpected logout error yesterday.

mdevaev, 2025-03-09:
Sup. Any updates?

mdevaev, 2025-03-12:
Closed due no feedback. Please update it if you want to continue or if you still have this problem.

closed 2025-03-12 by mdevaev