pikvm/pikvm #1472: Virtual trackpad between screen and keyboard where it's currently an empty area

Majestic7979, 2025-02-21:
Good day Maxim, I hope you're doing well :) So I opened the kvm on my iPhone and it works great, I didn't try before today to use on a mobile. I noticed that there is an area between the keyboard at the bottom and screen at the top that looks just like a trackpad... so I started sliding my finger on it thinking I could control the mouse just like using a laptop trackpad but nothing happened, while using the keyboard worked fine, so I thought something must be wrong. That's when I stopped for a coffee and realized it's NOT a trackpad lol... but then I started wondering why is it not a virtual trackpad? Seems like it would be a perfectly great addition because there's a keyboard and that space is currently blank, so when the user is on a smartphone and the phone is portrait orientation, then you get screen at the top, trackpad in the middle to move the mouse virtually, and keyboard at the bottom!!!! This would provide the user the ability to finely control the remote cursor on the tiny smartphone display which is a big win-win if you ask me... win two wins? You win by implementing a cool and useful feature and we win by being users of a developer who is such a great guy who listens to his users ;-)

Flattery aside (but it's honest) I wanted to also ask if you have considered doing web cam pass through from the host device to the remote device that the PiKVM is controlling... this is because I have a client who only allows me to work in one location, and I often have to have meetings with said client from this location. If I just took a photo of their room and used a virtual green screen I could then be remote, do the meeting with the virtual background using webcam pass through and they would probably not notice because everyone uses a virtual background on the meetings to avoid sensitive data leaking... so would be extremely awesome, and complement the microphone pass through you lovingly gave to us recently 🥰

mdevaev, 2025-03-09:
Hello!

It seems that the presence of this area is just a layout glitch. If my memory serves me correctly, you can use the entire stream area as a trackpad if you use relative mouse mode, and this is not possible for absolute.

Majestic7979, 2025-03-31:
I understand and yes, the screen can be used, but actually in a mobile it's tiny so if you could turn that area into an actual virtual trackpad, it would be possible to control very small UI elements that a fat finger can't precisely tap :) I hope you understand what I mean!! Clicking a "ok" button in a dialog in a tiny screen can sometimes feel like a humanly impossible task

Majestic7979, 2026-03-23:
Oh wow, thanks man! I will test this some time! Appreciate you always listening to customer needs, sending bestest wishes to you Maxim!

mdevaev, 2026-03-23:
👌