pikvm/pikvm #1473: Virtual PiKVM Flash Drive not reocgnized on Windows Host

vieacq, 2025-02-28:
**Describe the bug**
Virtual PiKVM Flash Drive not reocgnized on Windows Host.
Drive in Device Manager always shows as "PiKVM Optical Drive USB Device" under "USB Mass Storage Device"
**To Reproduce**
1. Select "Drive" in the web-GUI
2. Choose .img-file to be presented as content of the flash-drive
    - not an optical drive image
    - in this case a dd-generated empty .img (8GB)
4. Select "Flash" and "Writeable"
5. Select "Connect drive to Server"
6. In Windows Explorer: unformatted Drive error (expected)
    - should be used as target for rufus
    - in Device Manager: "PiKVM Optical Drive USB Device" under "USB Mass Storage Device"
 7. Unsuccessful:
    - Deinstall "PiKVM Optical Drive USB Device" & Search for new devices
    - Deinstall "PiKVM Optical Drive USB Device" & deactivate "USB Mass Storage Device" then reactivate
    - Reboot PiKVM
**Expected behavior**
A Virtual PiKVM Flash Drive should be rw-accessible provisioned

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 22H2
 - Browser: Edge (not relevant here)

**PiKVM info:**
  - PiKVM V3 Pre-Assembled
  - kvmd 4.62-1
  - kvmd-fan 0.32-1
  - kvmd-platform-v3-hdmi-rpi4 4.62-1
  - kvmd-webterm 0.50-1

**Additional context**
It seems that PiKVM boots with Drive in "CD/DVD" mode, disconnected.
Windows recognizes this as "PiKVM Optical Drive USB Device" with no media inserted
Subsequent change to Flash-Massstorage is not honored.

Suggestion:
PiKVM provisions both Virtual-USB devices ("Flash" and "CD/DVD") and then may connect the appropriate image to the respective device. Thus Windows can load and provision the correct drivers while the PiKVM OTG stays active.

mdevaev, 2025-02-28:
Did you try this? https://docs.pikvm.org/msd/

[image]

vieacq, 2025-02-28:
Thanks for the quick reply...

I have seen this and seemingly mixed that up with "Reset HID".

Now I did 
1. disconnect the USB
2. then setup the Flash and "connected" it
3. then reconnected the USB

SUCCESS: Windows recognized "PiKVM Flash Drive USB Device"

To me this seems a bit clumsy, so I hope this will be a bit more streamlined in the future - maybe on the lines I suggested in my previous comment.

Also, now I hope the Flash Drive setting (in PiKVM) survives reboot, because I plan to boot from the Flash Drive (rescue system).

closed 2025-02-28 by vieacq

mdevaev, 2025-02-28:
The setting is not saved after reboot, but if you want, you can set the flash mode by default using this override:

```yaml
otg:
    devices:
        msd:
            default:
                cdrom: false
```

As for the inconvenient mode switching, I'm working on it and will soon make a convenient tool to automate this action.