pikvm/pikvm #1474: make os fails with: rm: cannot remove '/etc/systemd/network/eth.network': No such file or directory

TheMachine001, 2025-03-06:
**Describe the bug**
When trying to build a pikvm image, the build stops with:
Step 30/94 : RUN rm /etc/systemd/network/en.network /etc/systemd/network/eth.network
 ---> Running in 83fa01d8adb4
rm: cannot remove '/etc/systemd/network/eth.network': No such file or directory
The command '/bin/sh -c rm /etc/systemd/network/en.network /etc/systemd/network/eth.network' returned a non-zero code: 1
make[1]: *** [Makefile:186: os] Error 1
make[1]: Leaving directory '/home/kevins/Downloads/os/.pi-builder/v2-hdmi-zero2w-arm'
make: *** [Makefile:61: os] Error 2

**To Reproduce**
Steps to reproduce the behavior, like:
1.  download pikvm git repository
2. create config.mk in the os folder
3. run make os

**Expected behavior**
I expected the pikvm os to be built without error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser firefox
 - Version 135.0.1
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version pi zero 2 WH
  - PiKVM platform v2-hdmi
  - Video capture type CSI bridge
  - KVMD version: image not yet built
  - uStreamer version: image not yet built
  - Linux kernel: image not yet built

**Additional context**

mdevaev, 2025-03-08:
Fixed. Please reclone the os repo.

closed 2025-03-08 by mdevaev

TheMachine001, 2025-03-09:
Hi @mdevaev,
"make os" completed successfully. Thank you very much for the fix.