pikvm/pikvm #1478: Commit last week broke v1 OS Image download

IllusionVK, 2025-03-17:
In the commit last week https://github.com/pikvm/pikvm/commit/b23fb02985b069e42e1f490d08229eb2e14321cb by @mdevaev   the link to the images was changed to be https://files.pikvm.org/images/v1-hdmi-rpi3/arm/v1-hdmi-rpi3-arm-latest.img.xz but this is now a 404

i found that the file is still accessible in its old location ? https://files.pikvm.org/images/v1-hdmi-rpi3/arm/v1-hdmi-rpi3-latest.img.xz

IllusionVK, 2025-03-17:
hmmmmmm.....  i just re-checked things and it looks like the problem is that https://docs.pikvm.org/flashing_os/ is not showing the changes checked into GitHub, because i just used the link from the change in github https://files.pikvm.org/images/v1-hdmi-rpi3/aarch64/v1-hdmi-rpi3-aarch64-latest.img.xz and it worked fine

mdevaev, 2025-03-17:
Fixed, thank you.

closed 2025-03-17 by mdevaev