pikvm/pikvm #1479: Disable USB mouse support directly in Pico HID

mclei-asw, 2025-03-19:
**Is your feature request related to a problem? Please describe.**
I have problem with a multiport Intellinet KVM. It seems it does not handle correctly the USB HID mouse. When I was trying an OTG, the keyboard started working only when I have disabled the mouse over OTG. But still, the keyboard works randomly. So I tried the Pico HID. But the keyboard also does not work at all. I think, it is because it advertises also a mouse device.
I know, that the Pico HID is working, because when I connect it directly to a computer, both keyboard and mouse works.

**Describe the solution you'd like**
I would like to be have a Pico HID firmware without an USB mouse support. Or to be able to disable it permanently like it is possible with PS/2 devices.

mdevaev, 2025-03-19:
Hello. The fastest way is enable PS/2 and enable PS/2 mouse preferring over USB. Connect both `GP2` and `GP4` pins to `GND`.

https://docs.pikvm.org/pico_hid/#configuring-the-hid-modes

stappersg, 2025-03-19:
On Wed, Mar 19, 2025 at 09:25:26AM -0700, Maxim Devaev wrote:
> The fastest way is enable PS/2 and enable PS/2 mouse preferring over
> USB. Connect both `GP2` and `GP4` pins to `GND`.
> 
> https://docs.pikvm.org/pico_hid/#configuring-the-hid-modes

From that URL:
|Pin name on the Pico board: 	Description
|GP2: 	Enable PS/2 keyboard & mouse support (see below).
|GP3: 	Prefer the PS/2 keyboard over USB when turning on the HID (if PS/2 is enabled).
|GP4: 	Prefer the PS/2 mouse over USB (if PS/2 is enabled)

Yes, I think that solves the original request of
"Disable USB mouse support directly in Pico HID"

Groeten
Geert Stappers
-- 
Silence is hard to parse

mdevaev, 2025-03-20:
Solved, I guess.

closed 2025-03-20 by mdevaev

mclei-asw, 2025-03-21:
Confirming, that it works this way any only USB Keyboard is advertised.

(Just it does not solve my compatibility problem with that KVM :-( )