pikvm/pikvm #1480: Blue Dot dissappears on Chrome fullscreen

FernandoM-eng, 2025-03-21:
**Describe the bug**
When using PIKVM on fullscreen in Chrome, if the option in the mouse menu "show blue dot" is enabled, the dot pointer is shown but after clicking the window context, it disappears making it difficult to control the remote machine.

**To Reproduce**
1. Connect to the pikvm with username and password
2. Go to the KVM option
3. Enable "show the blue dot" under the mouse settings
4. Click on the fullscreen icon within the pikvm interface. 
5. Click anywhere in the window and move the cursor.
6. See error**Expected behavior**
The blue dot disappears from the full screen view. Only the host screen with it's own pointer is shown.

**Screenshots**
No screenshots 

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - Version 134.0.6998.89

**PiKVM info:**
  - Raspberry Pi Zero W
  - PiKVM platform: DIYv2
  - Video capture type: CSI bridge
  - KVMD version: kvmd 4.66-1
kvmd-fan 0.32-1
kvmd-platform-v2-hdmi-zero2w 4.66-1
kvmd-webterm 0.50-1
  - uStreamer version: ustreamer 6.35-1
  - Linux kernel: Linux pikvm 6.6.45-11-rpi #1 SMP Sat Feb 22 20:43:12 UTC 2025 armv7l GNU/Linux

mdevaev, 2025-03-24:
Hello. Please try incognito mode. I can't reproduce it.

FernandoM-eng, 2025-03-24:
I tried in incognito and also using Edge browser (chromium based). Same thing. I'll update to win11 and maybe that will fix it.

If it cannot be reproduced, there's no point on having this issue open. I'll close it.

Thanks!!

mdevaev, 2025-03-29:
No, still can't reproduce :/ Did you try incognito mode?

FernandoM-eng, 2025-03-29:
Yes, I tried incognito mode. Must be something else. Anyway, direct mode works for me, I'll use it that way.
Same as before, if you can't reproduce, there's no point on keeping the issue open.

Thanks a lot for taking the time to try and reply! Keep on the amazing work!!!

closed 2025-03-29 by FernandoM-eng

mdevaev, 2025-03-29:
Maybe you're using the relative mouse mode?

FernandoM-eng, 2025-03-29:
No. These are the settings. 
[image]

The dot is shown, but after going fullscreen using the "go to fullscreen mode" button in the kvm interface, the dot disappears after a few seconds.

mdevaev, 2025-03-29:
If this is a host issue, I can't fix it :(

FernandoM-eng, 2025-03-29:
Thanks for giving it a try!