pikvm/pikvm #1481: PiKVM OS no longer ships platformio

smumriak, 2025-03-22:
**Describe the bug**
I'm using the old v2 platform that has arduino-based HID controller. Nowadays the firmware flushing is impossible because platformio is no longer preinstalled.

**PiKVM info:**
  - RPi 4
  - DIY v2

mdevaev, 2025-03-22:
You can manually install platformio with `pacman -S platformio-core`.

medicalwei, 2025-03-22:
I think it is the problem of documentation
https://docs.pikvm.org/flashing_hid/

I'll supply you a PR on the suggestions

smumriak, 2025-03-22:
@mdevaev I'll leave the decision to include this in documentation for legacy HID up to you at this point :) Feel free to close this issue, as installing it is the solution. Thanks!

medicalwei, 2025-03-22:
I found there are lots of problem other than just docs. The avrdude included in ArchLinux does not work according to doc due to the removal of sysfs access to GPIO, as well as the incompatible change of gpioset command arguments.

mdevaev, 2025-03-22:
Please show the log of avrdude usage.

medicalwei, 2025-03-23:
I opened the issue into #1482

PR #1483 (2025-03-23): docs: update flashing_hid.md
This changeset adds manual on installing platformio since recent image does not ship with it.  Also fix some typo due to directory name change.

Closes #1481
closed 2025-03-23 by mdevaev