pikvm/pikvm #1482: Several Arduino HID (using SPI) related firmware flashing issues

medicalwei, 2025-03-23:
**Describe the bug**
I cannot follow the documentation to flash Arduino HID firmware using SPI connection.  The hardware was made some years ago but the firmware failed to flash through SPI.

**To Reproduce**
1. Install PiKVM v2 firmware, have a Arduino HID connected using SPI interface,
2. ssh into Pi, `su -`, `rw`, `pacman -Syu`, `pacman -S platformio-core`
3. `cp -R /usr/share/kvmd/hid/arduino ~`, `cd arduino`
4. `make spi`
5. `make spi install`

**Expected behavior**
Firmware should be able to install to Arduino Micro connected to the Raspberry Pi

**Actual behavior**
Cannot communicate through GPIO due to the command change

```
[root@pikvm arduino]# make install
bash -ex -c " \
        current=`cat .current`; \
        if [ 'spi' == 'spi' ] || [ 'spi' == 'aum' ]; then \
                gpioset 0 25=1; \
                gpioset 0 25=0; \
        fi \
"
+ current=spi
+ '[' spi == spi ']'
+ gpioset 0 25=1
gpioset: invalid line value: '0'
make: *** [Makefile:24: upload] Error 1
```

If changing Makefile then update `gpioset` command into the following:
```
gpioset -c gpiochip0 -t 30ms,0 25=1; \
gpioset -c gpiochip0 -t 30ms,0 25=0; \
```
it can continue, then stuck at avrdude:

```
platformio run --environment 'spi' --project-conf 'platformio-avr.ini' --target upload
Processing spi (platform: atmelavr; board: micro; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/micro.html
PLATFORM: Atmel AVR (5.1.0) > Arduino Micro
HARDWARE: ATMEGA32U4 16MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) External (simavr)
PACKAGES:
 - framework-arduino-avr @ 5.2.0
 - tool-avrdude @ 1.70200.0 (7.2.0)
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- HID-Project @ 2.8.2+sha.afae388
|-- ps2dev @ 0.0.3+sha.a043002
|-- digitalWriteFast @ 1.0.0
|-- HID @ 1.0
|-- drivers-avr
|-- drivers
Building in release mode
Checking size .pio/build/spi/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  15.6% (used 399 bytes from 2560 bytes)
Flash: [====      ]  36.5% (used 10466 bytes from 28672 bytes)
Configuring upload protocol...
AVAILABLE: custom
CURRENT: upload_protocol = custom
Uploading .pio/build/spi/firmware.hex
avrdude error: Linux SPI driver not available in this configuration
avrdude error: programmer does not support open()
avrdude error: unable to open programmer rpi on port /dev/spidev0.0:/dev/gpiochip0

avrdude done.  Thank you.

*** [upload] Error 1
=== [FAILED] Took 3.05 seconds ===

Environment    Status    Duration
-------------  --------  ------------
spi            FAILED    00:00:03.052
=== 1 failed, 0 succeeded in 00:00:03.052 ===
make: *** [Makefile:25: upload] Error 1
```

I finally managed to flash the firmware using avrdude built with libgpiod support, however I don't think that is a upstreamable solution.

**PiKVM info:**
  - Raspberry Pi 4B with 4GB RAM
  - PiKVM v2 with USB capture card
  - KVMD version: 4.67-1
  - uStreamer version: 6.35-1
  - Linux kernel: Linux pikvm 6.6.45-11-rpi #1 SMP PREEMPT Sat Feb 22 18:52:52 UTC 2025 aarch64 GNU/Linux

mdevaev, 2025-03-23:
Um, how did you flash it if avrdude reports an error?

```
Environment    Status    Duration
-------------  --------  ------------
spi            FAILED    00:00:03.052
=== 1 failed, 0 succeeded in 00:00:03.052 ===
```

medicalwei, 2025-03-23:
I downloaded PKGBUILD from https://gitlab.archlinux.org/archlinux/packaging/packages/avrdude

Then modified it to include libgpiod as dependency and enable flag `-D HAVE_LIBGPIOD=1`, and build package with `makepkg`, finally I flashed the built binary using the command:

```
avrdude -C +avrdude-rpi.conf -p atmega32u4 -c rpi -P gpiochip0 -v -U flash:w:.pio/build/spi/firmware.hex
```

where I also modified `avrdude-rpi.conf` to:
```
programmer
        id = "rpi";
        desc = "RPi SPI programmer";
        type = "linuxgpio";
        prog_modes = PM_ISP;
        connection_type = linuxgpio;
        reset = 25;
        sck = 11;
        sdo = 10;
        sdi = 9;
;
```

mdevaev, 2025-03-23:
Please try the default avrdude from the repo but change `/dev/spidev0.0:/dev/gpiochip0` in platformio-avr.ini just to `/dev/spidev0.0`.

mdevaev, 2025-03-24:
Sup?

medicalwei, 2025-03-24:
Still here. I am physically away from the setup for 4 more days but I can try to remote into it.

medicalwei, 2025-03-25:
```
Uploading .pio/build/spi/firmware.hex
avrdude error: Linux SPI driver not available in this configuration
avrdude error: programmer does not support open()
avrdude error: unable to open programmer rpi on port /dev/spidev0.0

avrdude done.  Thank you.
```

Am getting the same error message.

medicalwei, 2025-03-25:
If I use avrdude in the repo instead of the one coming with platformio, it will use the config from platformio and spur error messages:
```
Uploading .pio/build/spi/firmware.hex
Warning: avr_mem_order[] does not know data; add to array and recompile
Error: unable to locate AVR XMEGA family common values's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:15527]
Error: unable to locate AVR XMEGA-E family common values's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:15544]
Error: unable to locate AVR XMEGA-A family common values's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:15565]
Error: unable to locate ATxmega16A4U's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:15635]
Error: unable to locate ATxmega16C4's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:15662]
Error: unable to locate ATxmega16D4's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:15691]
Error: unable to locate ATxmega16A4's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:15715]
Error: unable to locate ATxmega32A4U's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:15785]
Error: unable to locate ATxmega32C4's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:15812]
Error: unable to locate ATxmega32D4's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:15842]
Error: unable to locate ATxmega32A4's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:15866]
Error: unable to locate ATxmega64A4U's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:15935]
Error: unable to locate ATxmega32C3's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:15961]
Error: unable to locate ATxmega32D3's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:15988]
Error: unable to locate ATxmega64C3's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:16015]
Error: unable to locate ATxmega64D3's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:16043]
Error: unable to locate ATxmega64D4's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:16071]
Error: unable to locate ATxmega64A1's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:16096]
Error: unable to locate ATxmega64A1U's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:16120]
Error: unable to locate ATxmega64A3's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:16139]
Error: unable to locate ATxmega64A3U's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:16162]
Error: unable to locate ATxmega64A4's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:16202]
Error: unable to locate ATxmega64B1's data in multi-memory address space
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:16228]
[... cut ...]
Warning: revise fileio_mem_offset(), avrdude.conf entry or memory type assignment [/root/arduino/.platformio/packages/tool-avrdude/avrdude.conf:22251]
Warning: system wide configuration file version ()
does not match Avrdude build version (8.0)
Warning: programmer rpi fails to specify prog_modes = PM_...; [/root/arduino/avrdude-rpi.conf:7]
Error: programmer rpi and part ATmega32U4 have no programming modes in common
use -c? -p ATmega32U4 to see all possible programmers for ATmega32U4
*** [upload] Error 1
```

medicalwei, 2025-03-25:
Now I am getting a working combination using avrdude provided by repo:

Install avrdude along with platformio-core:

```
pacman -S avrdude platformio-core
```

Then change the configuration in avrdude-rpi.conf so it passes prog_modes required by avrdude 8.0:

```
programmer
	id = "rpi";
	desc = "RPi SPI programmer";
	type = "linuxspi";
	prog_modes = PM_ISP;
	reset = 25;
	baudrate = 400000;
;

```

Change Makefile so the gpioset command has correct parameters:

```
			gpioset -c gpiochip0 -t 30ms,0 25=1; \
			gpioset -c gpiochip0 -t 30ms,0 25=0; \
```

I can flash the Arduino Micro on board by having the above changeset.

Can file a PR later after work.

mdevaev, 2025-03-25:
Merged, thank you.

closed 2025-03-25 by mdevaev