pikvm/pikvm #1485: USB passthru

SemperFi89, 2025-03-25:
I request USB pass-thru from the PIKVM=>PiKVM Switch=>computer.  The main thing I'm looking for here is a keyboard/mouse that would control the selected computer.  I have the 4+ and would like the HDMI pass-thru to a local monitor with a local kb/mouse that could control the chosen computer.

If this can be done, the 2nd request would be for a hotkey combo to change the port the PiKvm switch controls.

I don't know if either is possible. Thanks.

FearNaBoinne, 2025-04-04:
I would love USB passthrough as well... But not for keyboard per sé. It would be nice if I could pass through a USB webcam, so I can join meetings remotely when I do NOT have my work laptop with me (Company has restricted access to any asset that requires MS Auth login, which is virtually everything, to registered computers only and disallows remote desktop applications like Anydesk, Teamviewer, etc.)

mdevaev, 2025-04-04:
I will add KB/Mouse passthrough soon.

StephenShamakian, 2025-05-15:
I just bought a PiKVM v4 Plus and made a huge mistake in assuming the keyboard/mouse would passthrough with the video when reading the docs on the site. So really would love this feature as well! :)

mdevaev, 2025-05-15:
@StephenShamakian keyboard/mouse passthrough will be available in this month or at the beginning of the next one.

StephenShamakian, 2025-05-15:
@mdevaev BTW it looks like Techno Tim on YouTube (https://youtu.be/NKSIpHNQCUA?si=XEkYNUq7Tp3xiCNb&t=336) was able to figure out a workaround until this can be implemented in software. But isn't really clear on what he means by upstream and downstream and if its on the CTL USB or the OTG USB. Would you happen to know how he has this hooked up to get this functionality working in the mean time? I have a PiKVM 4 Plus with a single switch. I've been trying a few different combinations without much luck using an externally powered USB 3.1 Hub.

mdevaev, 2025-05-15:
Downstream on the hub is a usb male connector to the host, you should connect it to otg upstream on the switch.

StephenShamakian, 2025-05-15:
@mdevaev hmm ok, tried that: USB hub's male connector plugged into Switch's Uplink OTG port, and PiKVM's OTG plugged into one of the ports on the USB hub.

But still no luck. Also in doing that I loose all USB control (even through the Web UI) tried a few different USB 3.1 hubs as well.

mdevaev, 2025-05-16:
You need a hub with external power because Switch doesn't give 5V on the line.

StephenShamakian, 2025-05-16:
Yeah I'm using this Powered USB hub (and plugging into the data ports): https://a.co/d/2GYcFYu

I'll try a few others to see if it's just a compatibility issue with this one. But regardless not a huge deal the official future software solution would be the best outcome anyway. 🙂

closed 2025-05-23 by mdevaev