pikvm/pikvm #1488: BUG: spinlock bad magic

floydelcy, 2025-03-29:
**Describe the bug**

After the microphone function is turned on，If unplugged the USB-OTG the kernel will show the BUG message

**To Reproduce**
Steps to reproduce the behavior, like:
1，enable MicroPhone function
2，Plug-out USB-OTG cable form target or turn off switch in GUI

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
```
[Thu Mar 27 06:16:16 2025] BUG: spinlock bad magic on CPU#0, irq/86-dwc3/1259
[Thu Mar 27 06:16:16 2025]  lock: 0xffffff80075ade60, .magic: 00000000, .owner: <none>/-1, .owner_cpu: 0
[Thu Mar 27 06:16:16 2025] CPU: 0 PID: 1259 Comm: irq/86-dwc3 Not tainted 5.10.160-rockchip-rk356x #1.0.6
[Thu Mar 27 06:16:16 2025] Hardware name: Rockchip RK3566 Orange Pi CM4 Board (DT)
[Thu Mar 27 06:16:16 2025] Call trace:
[Thu Mar 27 06:16:16 2025]  dump_backtrace+0x0/0x1b0
[Thu Mar 27 06:16:16 2025]  show_stack+0x20/0x2c
[Thu Mar 27 06:16:16 2025]  dump_stack_lvl+0xc8/0xf8
[Thu Mar 27 06:16:16 2025]  dump_stack+0x18/0x34
[Thu Mar 27 06:16:16 2025]  spin_bug+0x90/0xb4
[Thu Mar 27 06:16:16 2025]  do_raw_spin_lock+0x40/0xdc
[Thu Mar 27 06:16:16 2025]  _raw_spin_lock_irqsave+0x2c/0x40
[Thu Mar 27 06:16:16 2025]  set_active+0x30/0x78
[Thu Mar 27 06:16:16 2025]  u_audio_suspend+0x34/0x44
[Thu Mar 27 06:16:16 2025]  afunc_suspend+0x1c/0x28
[Thu Mar 27 06:16:16 2025]  composite_suspend+0xc4/0xd8
[Thu Mar 27 06:16:16 2025]  configfs_composite_suspend+0x78/0x7c
[Thu Mar 27 06:16:16 2025]  dwc3_suspend_gadget+0x4c/0x64
[Thu Mar 27 06:16:16 2025]  dwc3_thread_interrupt+0x86c/0x9d8
[Thu Mar 27 06:16:16 2025]  irq_thread_fn+0x34/0x7c
[Thu Mar 27 06:16:16 2025]  irq_thread+0x108/0x1b4
[Thu Mar 27 06:16:16 2025]  kthread+0xf4/0x104
[Thu Mar 27 06:16:16 2025]  ret_from_fork+0x10/0x30
```
**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
 ported to the rk3566,but i think there will be have the same issue on RPI

**Additional context**
The microphone functionality sets`c_chmask` to `0` during initialization, which causes `g_audio_setup` not to initialize its spinlock `(spin_lock_init(&prm->lock))`

```
drivers/usb/gadget/function/u_audio.c

int g_audio_setup(struct g_audio *g_audio, const char *pcm_name,
					const char *card_name)
{
...
if (c_chmask) {
	struct uac_rtd_params *prm = &uac->c_prm;
	spin_lock_init(&prm->lock);
```
However, in `u_audio_suspend`, 
```
void u_audio_suspend(struct g_audio *audio_dev)
{
	struct snd_uac_chip *uac = audio_dev->uac;

	set_active(&uac->p_prm, false);
	set_active(&uac->c_prm, false);
}
```
`u_audio_stop_capture`, 
```
void u_audio_stop_capture(struct g_audio *audio_dev)
{
	struct snd_uac_chip *uac = audio_dev->uac;

	set_active(&uac->c_prm, false);
	if (audio_dev->in_ep_fback)
		free_ep_fback(&uac->c_prm, audio_dev->in_ep_fback);
	free_ep(&uac->c_prm, audio_dev->out_ep);

	audio_dev->usb_state[SET_INTERFACE_OUT] = true;
	audio_dev->stream_state[STATE_OUT] = false;
	schedule_work(&audio_dev->work);
}
```
and `u_audio_start_capture`, there are calls to `set_active(&uac->c_prm, false/true)` 
```
static void set_active(struct uac_rtd_params *prm, bool active)
{
	// notifying through the Rate ctrl
	unsigned long flags;

	spin_lock_irqsave(&prm->lock, flags);
	if (prm->active != active) {
		prm->active = active;
		snd_ctl_notify(prm->uac->card, SNDRV_CTL_EVENT_MASK_VALUE,
				&prm->snd_kctl_rate_id);
	}
	spin_unlock_irqrestore(&prm->lock, flags);
}
```
with `spin_lock_irqsave` and `spin_unlock_irqrestore` when USB-OTG disconnected the kernel show the ` BUG: spinlock bad magic on CPU#0, irq/86-dwc3/1215` message

mdevaev, 2025-03-29:
Is it affected some functionality? I suppose you're right, and this is a bug in the audio module. But if that's how it works, then I'll look at it when I have time.

floydelcy, 2025-03-30:
No ,the issue doesn't affect the functionality, Only show a BUG message when you first unplugged the USB-OTG