pikvm/pikvm #1489: Add more host keymap options

alethiophile, 2025-03-30:
**Is your feature request related to a problem? Please describe.**
The server I run PiKVM against is set to a Colemak keyboard layout. I have no desire to change this, since I type Colemak instinctively and I also keep a physical keyboard plugged in to that server.

This is fine when simply typing into the KVM, but it messes up pasting. Any text pasted through the interface ends up going through a reverse Colemak->QWERTY mapping, which obviously makes it unusable (and the odd nature of this mapping makes it a pain to type the appropriate text manually).

The list of keymaps in the host keymap selection for pasting includes various different international keymaps, but none of the common variations such as alternate keymaps for the same language and nationality. Thus, I can't simply select Colemak in that listing.

**Describe the solution you'd like**
I would like Colemak (and, ideally, other common variant mappings such as Dvorak) included in the host keymaps listing.

Code-wise, this is trivial. I have solved this problem locally with:

```
$ sudo pacman -S qemu-tools
$ qemu-keymap -v colemak >en-us-colemak
$ scp en-us-colemak root@pikvm:/usr/share/kvmd/keymaps
```

This makes the Web interface pick up the new keymap with no further work. All that would be required to ship this would be including that colemak keymap in the distribution.

alethiophile, 2025-03-30:
For convenience, I attach here the file that was generated by `qemu-keymap` using the above invocation:

[en-us-colemak.txt](https://github.com/user-attachments/files/19527552/en-us-colemak.txt)

mdevaev, 2025-04-05:
Hello! Thank you, I've added it to 4.70 release.

closed 2025-04-05 by mdevaev