pikvm/pikvm #1490: Unable to update due to pacman error - cannot read db extra

Majestic7979, 2025-03-31:
pikvm-update
+ trap on_error ERR
+ _yes='--noconfirm --ask=4 --overwrite \*'
+ rw
+ mount -o remount,rw /
+ mount -o remount,rw /boot
+ set +x
=== PiKVM is in Read-Write mode ===
+ pacman -Syy
:: Synchronizing package databases...
 core                                                                   238.9 KiB   405 KiB/s 00:01 [###########################################################] 100%
 extra                                                                    9.3 MiB  16.1 MiB/s 00:01 [###########################################################] 100%
 community                                                               45.0   B   900   B/s 00:00 [###########################################################] 100%
 alarm                                                                   95.7 KiB  1063 KiB/s 00:00 [###########################################################] 100%
 aur                                                                     11.1 KiB   185 KiB/s 00:00 [###########################################################] 100%
 pikvm                                                                   11.6 KiB  33.2 KiB/s 00:00 [###########################################################] 100%
error: failed retrieving file 'core.db' from mirror.osbeck.com : The requested URL returned error: 404
error: failed retrieving file 'extra.db' from mirror.osbeck.com : The requested URL returned error: 404
error: failed retrieving file 'community.db' from mirror.osbeck.com : The requested URL returned error: 404
warning: too many errors from mirror.osbeck.com, skipping for the remainder of this transaction
++ pacman -S -u --print-format %n
++ grep -v pikvm-os-updater
++ wc -l
error: could not read db 'extra' (Damaged tar archive (bad header checksum))
error: could not read db 'extra' (Damaged tar archive (bad header checksum))
error: could not read db 'extra' (Damaged tar archive (bad header checksum))
error: could not read db 'extra' (Damaged tar archive (bad header checksum))
error: could not read db 'extra' (Damaged tar archive (bad header checksum))
error: could not read db 'extra' (Damaged tar archive (bad header checksum))
error: could not read db 'extra' (Damaged tar archive (bad header checksum))
error: could not read db 'extra' (Damaged tar archive (bad header checksum))
error: could not read db 'extra' (Damaged tar archive (bad header checksum))
error: could not read db 'extra' (Damaged tar archive (bad header checksum))
error: could not read db 'extra' (Damaged tar archive (bad header checksum))
error: could not read db 'extra' (Damaged tar archive (bad header checksum))
error: could not read db 'extra' (Damaged tar archive (bad header checksum))
error: could not read db 'extra' (Damaged tar archive (bad header checksum))

mdevaev, 2025-03-31:
Hello! Here the solution:
```
rw
pacman -U https://files.pikvm.org/tmp/libarchive-3.7.7-1-armv7h.pkg.tar.xz
pikvm-update
```

Majestic7979, 2025-03-31:
Thank you, it's fixed.

While you're here, if you don't mind Maxim, I need to ask your advice please.

I purchased this item:
https://www.aliexpress.com/item/1005006209126315.html?spm=a2g0o.order_list.order_list_main.11.12141802lDXoxI

I wired the NO contact and the COMMON contact to a switch in a normal KVM, this KVM has a button in front of it that when clicked, switches between input 1 and input 2, so I just soldered wires to the contacts and when the wires are shorted for a moment the KVM switches between the inputs. So I tried to follow the guide on pikvm.org for optocoupler (https://docs.pikvm.org/gpio/) using GPIO but it's not switching... could you please kindly advice/walk me through? Here's my override currently:

```
kvmd:
    hid:
        jiggler:
            enabled: true
            active: true
    gpio:
        drivers:
            otgconf:
                type: otgconf
            relay:
                type: hidrelay
                device: /dev/hidg0  # The path to the linux device
        scheme:
            relay1:  # Channel 1 of the relay /dev/hidraw0
                driver: relay  # Not GPIO, so add name from the above Drivers section
                pin: 0  # Numerating starts from 0
                mode: output  # Relays can't be inputs
                switch: false
                #initial: null  # Don't reset the state to 0 when initializing and terminating KVMD
                pulse:
                    delay: 0.5  # Default pulse value
                    max_delay: 0.5  # The pulse interval can be between min_delay=0.1 (by default) and max_delay=2
            hid.usb0:
                driver: otgconf
                mode: output
                pin: hid.usb0
                pulse: false
            hid.usb1:
                driver: otgconf
                mode: output
                pin: hid.usb1
                pulse: false
            mass_storage.usb0:
                driver: otgconf
                mode: output
                pin: mass_storage.usb0
                pulse: false
        view:
            header:
                title: Functions
            table:
                - []
                - ["#KVM 1/2:", "relay1|Switch Machine"]  # Text label, one input, one button with text "Click"
                - []
                - ['#Keyboard', '#hid.usb0', hid.usb0]
                - ['#Absolute Mouse', '#hid.usb1', hid.usb1]
                - ['#Mass Storage Drive', '#mass_storage.usb0', mass_storage.usb0]
 ```

Majestic7979, 2025-03-31:
```
ls /dev
block     v4l            hidg1               loop-control  mmcblk0    ram14      stderr  tty19  tty32  tty46  tty6       v4l-subdev0  vcsa4        video11    zero
bus       autofs         hidg2               loop0         mmcblk0p1  ram15      stdin   tty2   tty33  tty47  tty60      vc-mem       vcsa5        video12
char      btrfs-control  hwrng               loop1         mmcblk0p2  ram2       stdout  tty20  tty34  tty48  tty61      vchiq        vcsa6        video13
disk      cachefiles     i2c-0               loop2         mmcblk0p3  ram3       tty     tty21  tty35  tty49  tty62      vcio         vcsm-cma     video14
dma_heap  cec0           i2c-1               loop3         mmcblk0p4  ram4       tty0    tty22  tty36  tty5   tty63      vcs          vcsu         video15
dri       cec1           i2c-10              loop4         null       ram5       tty1    tty23  tty37  tty50  tty7       vcs1         vcsu1        video16
fd        console        i2c-20              loop5         port       ram6       tty10   tty24  tty38  tty51  tty8       vcs2         vcsu2        video18
input     cuse           i2c-21              loop6         ppp        ram7       tty11   tty25  tty39  tty52  tty9       vcs3         vcsu3        video19
mapper    fb0            i2c-22              loop7         ptmx       ram8       tty12   tty26  tty4   tty53  ttyAMA0    vcs4         vcsu4        video20
mqueue    full           kmsg                media0        ram0       ram9       tty13   tty27  tty40  tty54  ttyS0      vcs5         vcsu5        video21
net       fuse           kvmd-hid-keyboard   media1        ram1       random     tty14   tty28  tty41  tty55  ttyUSB0    vcs6         vcsu6        video22
pts       gpiochip0      kvmd-hid-mouse      media2        ram10      rfkill     tty15   tty29  tty42  tty56  ttyprintk  vcsa         vga_arbiter  video23
serial    gpiochip1      kvmd-hid-mouse-alt  media3        ram11      rtc        tty16   tty3   tty43  tty57  uhid       vcsa1        vhci         video31
shm       gpiomem        kvmd-video          media4        ram12      rtc0       tty17   tty30  tty44  tty58  uinput     vcsa2        video0       watchdog
snd       hidg0          log                 mem           ram13      spidev0.0  tty18   tty31  tty45  tty59  urandom    vcsa3        video10      watchdog0
```

Majestic7979, 2025-03-31:
```
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 003: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

Majestic7979, 2025-03-31:
```
ls /dev/serial/by-id
usb-1a86_USB_Serial-if00-port0
```

Majestic7979, 2025-03-31:
Ah =/ Could you please recommend a link? Cheap as possible and that works well. 
Thanks.

closed 2025-04-05 by mdevaev