pikvm/pikvm #1491: Support Netbird as an alternative to Tailscale

B08Z, 2025-04-02:
**Netbird Alternative**
I like Tailscale but would prefer to use open source alternatives, headscale does provide an option but there - are still limitations in the namely not using native kernel Wireguard, among others. 

**The solution I'd like**
I would like to use Netbird as an alternative to Tailscale. Installed in the same way and enabled in a similar manner to tailscale-pikvm. This would enable the same / similar functionality but with netbird providing the Mesh VPN. 

**Alternatives I've considered**
Tailscale client with Headscale co-ord server. Works as expected but not fully open source or self hosted and lacks policy compliance like OS check etc. 
Wireguard standalone works perfectly but requires network configuration changes when deploying pikvm to sites and can be blocked if firewall are restricted. (Netbird will work on port 443 only if configured correctly)

**Additional context**

https://github.com/netbirdio/netbird
NetBird.io

Happy to test if anyone can make suggestions on how to enable. 

Netbird does install but states command not found then I try and run it.

arch1mede, 2025-04-10:
Hi, I ran through the installer from: https://github.com/physk/netbird-installer
It spit out a bunch of failures but it did install
I just ran `netbird up` afterwards and it assigned an IP but as far as I can tell, this isn't any better than Tailscale.

However with that said, I think this should be regulated to other 3rd party app's and should be solely done by the end user.

mdevaev, 2025-04-10:
I suppose the question can be considered answered. Since PiKVM OS is generic Linux, the installation of any software remains on the conscience of the user. We use Tailscale because we have experience with it, but if you want something special, you can try to do it. The above is a great start.

closed 2025-04-10 by mdevaev

B08Z, 2025-04-12:
Thanks for your comments and initial support.

I think Netbird has a few advantages (FOSS, users the native WG kernel and can implement Rosenpass)

But you are correct it essentially offers the same as tailscale. 

My issue is that with the RO filesystem netbird will not start at boot or work as intended. I am not sure it tailscale-pikvm implements something to fix this for tailscale and so the ask is could the same be done for netbird.

Thanks anyway for looking.

mdevaev, 2025-04-12:
Tailscale can work on RO, and its dev team made some efforts to do this.
Our package just contains some minor fixes for startup scripts. I didn't
check how netbird works, but your can make some research itself. If a small
tuning is enough, we can add this to our package and/or to the doc. In
other cases this issue should be addressed to the netbird team to add RO
support.

Best regards,
Max Devaev