pikvm/pikvm #1493: GPIO Pulse Does Not Return to Initial State for Active-Low Relay in PiKVM

radu-boboc, 2025-04-03:
**Describe the bug**

When using the GPIO configuration in PiKVM to control an active-low relay, the relay does not return to its initial state (HIGH, relay off) after a pulse is triggered. Instead, after the first click on the "Button" or "Pulse" element in the web interface, the relay remains activated (pin stays LOW, relay on). Subsequent clicks trigger a pulse that temporarily deactivates the relay (pin goes HIGH), but then the relay reactivates (pin returns to LOW). The expected behavior is for the relay to activate temporarily (pin goes LOW) and then return to its initial state (pin HIGH, relay off) after each click.

**To Reproduce**

Configure the GPIO in /etc/kvmd/override.yaml as follows:
kvmd:
  gpio:
    drivers:
      atx:
        type: gpio
        pins:
    scheme:
      relay_1:
        driver: atx
        pin: 5
        mode: output
        initial: true
        invert: true
        switch: false
        pulse:
          delay: 0.2
          active: false
    view:
      header:
        title: Control ATX
      table:
        - ["#Reset Relay1", "relay_1", "Button"]

Restart the PiKVM service:
systemctl restart kvmd

Open the PiKVM web interface and go to the "Control ATX" menu.
Click on the "Reset Relay1" button.
Observe that the relay activates (pin goes LOW) but does not return to HIGH (relay remains on).
Click the button again, and observe that the relay deactivates temporarily (pin goes HIGH for 0.2 seconds), then reactivates (pin returns to LOW).

**Expected behavior**

At startup, the relay should be off (pin HIGH, since the relay is active-low, with initial: true and invert: true).
When clicking the "Reset Relay1" button, the relay should activate temporarily (pin goes LOW for 0.2 seconds, as defined by pulse: {delay: 0.2, active: false}), then return to its initial state (pin HIGH, relay off).
Each subsequent click should repeat this behavior: activate the relay for 0.2 seconds, then deactivate it.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 
 - Browser Vivaldi

**PiKVM info:**
Raspberry Pi board version: Raspberry Pi 4
Video capture type: CSI bridge
kvmd 4.52-1
kvmd-fan 0.32-1
kvmd-platform-v2-hdmi-rpi4 4.52-1
kvmd-webterm 0.50-1

 uname -a
Linux pikvm 6.6.45-10-rpi #1 SMP Fri Jan 24 03:51:01 UTC 2025 armv7l GNU/Linux

**Additional context**
The relay is active-low, meaning it activates when the GPIO pin is LOW (0) and deactivates when the pin is HIGH (1). This is why invert: true is used in the configuration.
The issue persists even when using different interface types in the view section (e.g., "Pulse", "Button", or "Click").

Please help me fix the logic, because  I don't wanna keep the relay always on. I want it always off, except when it's doing a pulse.

mdevaev, 2025-04-05:
Hello. You have a configuration error: `invert: true` is wrong, it should be `inverted: true`.

radu-boboc, 2025-04-06:
you are correct, now everything works ok.
Thank you!!

closed 2025-04-06 by radu-boboc

mdevaev, 2025-04-06:
👌

anshtiwari314, 2025-11-02:
i have the similar issue with GPIO pins if i set mode: output & try to switch the button from web ui it doesn't toggle GPIO mode from HIGH to LOW. 
 
current problem ( GPIO pin output is always HIGH & not changing  )
similar when u do mode: input ( GPIO pin output is always LOW & not changing by clicking the button )

here is my override.yaml 

```
kvmd:
     gpio:
         drivers:
             my_relay:
                 type: gpio
         view:
             header:
                 title: Switches  # The menu title
             table:  # The menu items are rendered in the form of a table of text labels and controls
                 - ["#Generic GPIO leds"]  # Text starting with the sharp symbol will be a label
                 - []  # creates a horizontal separator and starts a new table
                 - ["#Button 1:", button1]  # Text label, one input, one button with text "Click"
                 - ["#Button 2:", button2]
                 - ["#relay3:", relay3 ]
                 - ["#Led4:", led4 ]
         scheme:
             # A certain device sends signals to the RPi and we want the PiKVM to display this as an led
             button1:
                 pin: 5 # GPIO pin number on the RPi
                 mode: output
                 initial: true
                 inverted: true
                 switch: true
                 pulse:
                     delay: 2
                     max_delay: 2
             button2:
                 pin: 6 # GPIO pin number on the RPi
                 mode: output
                 switch: true
             relay3:
                 driver: my_relay
                 pin: 12
                 mode: output
                 initial: null
                 inverted: true
             led4: 
                 pin: 13
                 mode: input
```

i m also sharing output of my http://pikvm_ip/api/gpio

```
{
    "ok": true,
    "result": {
        "model": {
            "scheme": {
                "inputs": {
                    "led4": {
                        "hw": {
                            "driver": "__gpio__",
                            "pin": "13"
                        }
                    }
                },
                "outputs": {
                    "button1": {
                        "hw": {
                            "driver": "__gpio__",
                            "pin": "5"
[... 110 lines cut ...]
            },
            "outputs": {
                "button1": {
                    "busy": false,
                    "online": true,
                    "state": false
                },
                "button2": {
                    "busy": false,
                    "online": true,
                    "state": false
                },
                "relay3": {
                    "busy": false,
                    "online": true,
                    "state": true
                }
            }
        }
    }
}

```

@mdevaev @radu-boboc can u pls guide me what i m doing wrong . Thankyou

anshtiwari314, 2025-11-02:
@radu-boboc  Thanks for replying , but the issue has now been resolved 

the problem i using 
mode: input ( web ui cannot toggle this bcz its a readable mode for signals in pi )

fixed 
mode: ouput ( 0v & 3.3 v ) ( u can only control the ouput mode from web ui which can be 0v or 3.3 v )

why i m using input mode , bcz i was tring to control a relay module 
bought from https://robu.in/product/5v-4-channel-relay-module/ ( building a smart IP powerstrip for cheap ATX  which will also work in devices which doesn't have atx pins in motherboard )

there are 6 pins in relay module , ( vcc , v1 , v2 , v3 , v4 , gnd ) 

i was powering the relay module 5v directly from pi 5v pin 
v1 connected to gpio5 
v2 connected to gpio6
v3 connected to gpio12
v4 connected to gpio13
 & gnd connected to pi gnd 

when u modified override.yaml with mode:output relay gets turned on  ( & get turned off when using mode:input  ) 
its a output based relay module its get activated when u connect signal pins to gnd ( don't know exactly how i was getting expected output from relay module while directly connected to raspberry pi )

but this behaviour u can't have with web ui , 
web ui only work with mode: ouput  ( either 0v or 3.3v )

i will use octocupler module for further setup this ...

[image]