pikvm/pikvm #1495: Microphone selection

mdevaev, 2025-04-08:
Subj. We need a menu or something like that to select the input audio device in the web ui.

mdevaev, 2025-05-28:
Hint: https://github.com/meetecho/janus-gateway/blob/master/html/demos/devices.js#L29 `listDevices()`

`_Janus.listDevices((devices) => __logInfo(devices), {"video": false, "audio": true});`.

cRoCx, 2025-07-01:
I would also be happy to get this feature, but on the other side there's so much software out there without offering a microphone selection, but instead relying on the users having their default communication device properly set in sound settings at the operating system level. Wouldn't call this a must-have feature to get an additional application level override of what the OS already provides. Maybe a clarification in the docs would suffice (if not already there).