pikvm/pikvm #1497: Feature Request: UVC Gadget Emulation for Client Webcam (like Audio) - Bounty Offered

0-don, 2025-04-11:
**Is your feature request related to a problem? Please describe.**

Cannot use client-side webcam in applications (like Teams/Zoom) on the host machine via PiKVM, especially when host software install is forbidden. The existing audio feature works great, but video calls need native camera access too.

**Describe the solution you'd like**

Implement **UVC (USB Video Class) gadget emulation** for PiKVM V3/V4. This would mirror the current UAC audio feature: receive client webcam video via WebRTC and present it as a standard USB webcam to the host via the OTG port, requiring no host drivers/software.

**Describe alternatives you've considered**

*   **Software (RDP/USB-over-IP):** Requires host installation (not allowed).
*   **Hardware Chains (NDI->HDMI->USB Capture):** Complex, likely still needs UVC gadget support on PiKVM.
*   **Screen Sharing:** Poor quality/compatibility for video calls.

**Additional context**

*   Builds directly on the successful UAC audio implementation: `https://docs.pikvm.org/audio`
*   Related to prior interest: Issue #741
*   **Bounty Offer:** This feature is highly valuable. **I offer a bounty/donation** specifically to support developing this UVC gadget emulation. Happy to discuss details.

mdevaev, 2025-05-03:
Hello. We plan to do this, but not with high priority since we have many queued tasks.

chenglidev, 2025-07-04:
@mdevaev I need this feature too. Anything that I can help?

mdevaev, 2025-07-04:
We're working on it. It will be available in this year.

noname9312, 2026-01-15:
> It's done almost on 50%.

Is there an expected date when we can expect it?

mdevaev, 2026-01-15:
Q1 2026

AdeAdeoti2016, 2026-08-20:
> No, not today, in few days.
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Hi..The thread is gone, how do we get the update? I've been waiting for
this for a while and that's why I joined your membership and have been
donating.

mdevaev, 2026-09-04:
For some reason this ticket was disappeared from github. Now it's here again.

The camera was released already: https://docs.pikvm.org/camera

AdeAdeoti2016, 2026-09-05:
Yeah, I was able to test it out. It worked well when I tested it out at
home simulating being away while my remote laptop is connected to my phone
hotspot. But when I actually traveled, it didn't work well, the video
stream kept going in and out or fading out.  Do you know how to fix this?
Here's what I've tried so far.

- Ruled out power/thermal — throttled=0x0, ~50°C.
- Read the streamer logs: constant "keyframe requested by sink client" =
clients losing sync.
- Checked Tailscale — connections were direct on the same LAN, so ruled out
relay bottleneck.
- Identified multiple clients hammering one encoder as a factor → test with
a single client.
- I changed H.264. GOP from 0 to 60 in the multimedia settings (0 made
every frame a keyframe), and tuned bitrate (5000 → 4000). This took it from
constant dropouts to occasional dropouts and self-recovering freezes
Any suggestions on the fix?

AdeAdeoti2016, 2026-09-05:
> We will start testing within a week.

I also free endpoints by disabling mass storage and the alt/relative mouse
because I don't' use any of them to free up some bandwidth.

mdevaev, 2026-09-05:
> stream kept going in and out or fading out

Camera stream on screen stream? It looks like it’s just a poor internet connection, and there’s not much we can do about it.

> Read the streamer logs: constant "keyframe requested by sink client" = clients losing sync.

This is one of the signs of a connection problem.

> I changed H.264. GOP from 0 to 60 in the multimedia settings (0 made every frame a keyframe)

It’s the opposite. 0 means there are no keyframes unless necessary; this helps save bandwidth. Each frame is a keyframe only on 1.