pikvm/pikvm #1498: Suspend video stream when browser window is not focused

dionorgua, 2025-04-14:
**Is your feature request related to a problem? Please describe.**

For 'long lived' sessions it makes sense to somehow "suspend" video streaming to save power and CPU usage on receiver side. 

**Describe the solution you'd like**
Some setting like "Suspend video after XX seconds of inactivity". Theoretically browser can detect 'tab is not focused' event.
Ideally stream should automatically resume when tab gains focus again. 

**Describe alternatives you've considered**
Maybe alternative is to disable/close whole browser tab..  But again. It's less user friendly

NiKiZe, 2025-04-14:
A big use case for KVM solutions is monitoring, in some cases there could be 4, 9 or even more simultaneously visible KVM sessions without focus, this doesn't mean I don't want to keep seeing them.
However in this scenario 1 or even 0.1 FPS update frequency would be useful if it could save power or resources. (Decoding of non changing stream on the client is probably not the big issue here?)

mdevaev, 2025-05-03:
@NiKiZe Changed or not, it's steal stream. A lower frame rate will definitely consume less resources, but FPS is regulated on the PiKVM side and cannot be changed for just one client.

@dionorgua Right now, you can close the stream window and no traffic will be consumed. This is a manual solution, but isn't it enough?

[image]

dionorgua, 2025-05-04:
@mdevaev Yeah. I know about this feature. Personally I don't see any reason why PiKVM have this built-in 'window manager' at all. Maybe you've plans to somehow show multiple windows at the same time (so that multiple PiKVM devices can work together)?

So if PiKVM will not ask for password, closing that stream window is basically exactly same as just closing browser tab: still mostly same effort is needed to open it again. And maybe even worse, opening that window again is mouse-only action.

My original motivation was to have PiKVM window always opened on virtual desktop and quickly switch to it when I need to test something on target machine. Ideally using keyboard only and as fast as possible.

At the same time I understand that this may be corner case. And we may 'generalize' it to just 'slow down' video stream to 1-0.1 FPS for 'monitoring'.

mdevaev, 2025-05-05:
Well, as I said, it's impossible to reduce fps just for one user. So the only solution is stopping the stream on inactivity.

https://developer.mozilla.org/en/docs/Web/API/Document/hidden

closed 2025-06-11 by mdevaev