pikvm/pikvm #1500: Confirm text paste using keyboard shortcut

DamianPala, 2025-04-22:
**Is your feature request related to a problem? Please describe.**
It will be very handy to have a keyboard shortcut to easily paste a text that you wrote in Text window. Currently there is no way to press `Paste` button using keyboard. Enter gives you a new line.

**Describe the solution you'd like**
To be consistent with other tools, the shortcut should be a combination `Ctrl + Enter`

**Describe alternatives you've considered**
`Ctrl + Shift + V` may be used as well.

mdevaev, 2025-05-03:
Yea, why not. It will be available in the next release. I will add `Ctrl+Enter`.

DamianPala, 2025-05-03:
Great! Thank you! Please test this feature in most common browsers browser to make sure this key combination is compatible or I can test it for you.

DamianPala, 2025-05-06:
I have another idea that will be really helpful too. Shortcut to open the Text window where you type your text to paste into the device. Ctrl + Shift + Alt + T seems to be a good choice. Ctrl + T opens a new tab, with shift opens from history, with Ctrl + Alt + T opens terminal on Ubuntu. Having both shortcut you can paste a text without using the mouse at all.

mdevaev, 2025-05-18:
I'll add a hotkey for that later. Quick paste is available in release 4.72, please upgrade using the `pikvm-update`.

closed 2025-05-18 by mdevaev