pikvm/pikvm #1510: pikvm not accesable on local network

lukee12zakk, 2025-04-30:
**Describe the bug**
When I do a fresh installation of pikvmOS, everything works fine. But if I install and setup tailscale with an exit node, it no longer is accessible on the local network. 

**To Reproduce**
1. install tailscale and configure.
2. set up an exit node
3. try to access it on local net

**Expected behaviour**
It used to be that it would be accesable but after issues with my tailscale account and not being able to add a different one and reinstalling, it broke
**Screenshots**
I'm not sure what to provide here

**Desktop (please complete the following information):**
 - OS: Ubuntu(latest non LTS), Windows 11 pro 23h2, android(one UI 6.1)
 - Browser: Zen Browser/Firefox, Chrome(android)
 - Version: Latest of everything
 - VNC client: None

**PiKVM info:**
  - Raspberry Pi board version Pi 4
  - PiKVM platform Geekworm a8
  - Video capture type: not sure
  - KVMD version: image 1
  - uStreamer version: image 2
  - Linux kernel: image 3

**Additional context**
Add any other context about the problem here.

[image]

[image]

[image]

closed 2025-04-30 by mdevaev

lukee12zakk, 2025-04-30:
Hello. The A8 uses the same hardware as the original ones. i also have a software issue and not a hardware issue, in case that is what you don't have support for.

Thanks

mdevaev, 2025-04-30:
No, it's not the same hardware.

PiKVM is a complete product in two versions: DIY and original hardware. Our support is working on the product. So if someone has produced and sold you hardware without burdening themselves with supporting the entire product, we can't help you with it. Please contact the manufacturer.