pikvm/pikvm #1514: Provide config option to --ignore=network-interface at boot.

xuyenvuong, 2025-05-07:
**Is your feature request related to a problem? Please describe.**
The `systemd-networkd-wait-online.service` should not wait for unused interface at boot.

**Describe the solution you'd like**
Mounting SAMBA at boot with `fstab` is slowing down the boot process to several minutes due to `systemd-networkd-wait-online.service` keep waiting and then failed on an unused network-interface.

The cifs mount option `x-systemd.automount` would trigger `systemd-networkd-wait-online.service` and slow down the system boot up to 5 mins. E.g.
```bash
`//192.168.x.x/contents/software/isos /var/lib/kvmd/msd/isos cifs ro,vers=3.1.1,noauto,x-systemd.automount,x-systemd.device-timeout=10,credentials=/root/.smbcredentials 0 0`
```
**Describe alternatives you've considered**
My current solution was:

```bash
# Disable any unused interface to speed up boot.
systemctl edit systemd-networkd-wait-online.service

[Service]
Type=oneshot
ExecStart=
ExecStart=/usr/lib/systemd/systemd-networkd-wait-online --ignore=wlan0
RemainAfterExit=yes
```

I wish there's a way to turn off unused network-interface for network-shared mount.

**Additional context**
Add any other context or screenshots about the feature request here.

mdevaev, 2025-05-09:
Hello. What do you mean at "unused" interface? Not configured at all, or configured, but not connected?

xuyenvuong, 2025-05-09:
Hi @mdevaev , 

Sorry for not being unclear. 

I have a DIY Pi-KVM that connects to WLAN with no problem. Once I try to config load ISO files from SAMBA server via `fstab`, without using `x-systemd.automount`, the mount point won't mount at all. To work around, I have to use `x-systemd.automount` in the `cifs` config option. 

During boot.  the `systemd-networkd-wait-online.service` would time out and error out after 4-5 mins (from the log `jounalctl -b`). Turn out the unused network interface `eth0` was the culprit even if its status was DOWN. By using `--ignore=eth0`, the boot was back to normal 30-40s range.

It took a bit of time to debug this issue for a novice user like myself. So, I wish that we have an option to --ignore the unused network-interface or at the least document it for others.

Thanks.

P.S. I also try with `systemd` solution as the work around, but it ran into the same issue. Once I ignore the `eth0` network interface, both solution: 1) Mount at boot via `fstab` and, 2) Mount via `systemd` were working fine.

mdevaev, 2025-05-11:
Got it. I'll answer later when we discuss this problem here.

intelfx, 2025-06-02:
@xuyenvuong If you still have access to the Pi-KVM with this network setup, could you please describe that network setup in more detail?

In this configuration (Wi-Fi is used for connectivity; Ethernet is unused but networkd trips over it), do you have the Ethernet physically plugged or unplugged? Despite Ethernet being unused, do you still have the networkd configuration file for it present, or have you removed the relevant configuration file completely?

What do `networkctl list`, `networkctl status`, `networkctl status wlan0` and `networkctl status eth0` say (substitute wlan0 and eth0 if those aren't your actual interface names)? If the configuration for the Ethernet interface is present (previous question), could you please post it as well (e.g., `networkctl cat @eth0`)?

intelfx, 2025-06-04:
@xuyenvuong Okay, the easiest thing you can do here is append `--any` to the s-n-w-o command line:

```
[Service]
ExecStart=
ExecStart=/usr/lib/systemd/systemd-networkd-wait-online --any
```

From cursory testing, this should properly ignore tailscale interface as well and only consider eth0 and wlan0. Could you please test this as well and see if it works properly with all combinations of network connectivity (i.e., succeeds if either eth0 or wlan0 is online, fails if neither are)? We'll add this configuration to the OS images then.

xuyenvuong, 2025-06-04:
Hi @intelfx,

In fact, I have had tried the `--any` option before too, and it was the same issue during boot. The only solution for me was to use `--ignore=eth0` option.

dragon788, 2025-06-07:
Have you added `_netdev` to your fstab options for the smb/cifs line so mount is smarter and doesn't try to load it until the network is available?

xuyenvuong, 2025-06-10:
OK here is my test.

First, I added `_netdev` option to the `/etc/fstab` file.
Next, change to `ExecStart=/usr/lib/systemd/systemd-networkd-wait-online --any` and remove `RemainAfterExit=yes`

Then execute `reboot` command.

System boot up, but `wlan0` interface got these errors from the direct console, and without any DHCP assigned `ip`. I reboot several more times and it is still the same issue.

[image]

Next, I proceed to roll back the above changes, and `wlan0` interface instantly working after reboot.

What other configuration you guys want me to test? I am happy to help doing the unit testing for you.

intelfx, 2025-06-29:
@xuyenvuong Apologies for protracted silence; we've been pretty busy with other stuff.

I was not able to reproduce this issue when I first gave the `--any` advice. Looking at the screenshotted log now, I see that your Wi-Fi setup fails to establish connectivity on the link layer (CTRL-EVENT-ASSOC-REJECT).

It might be something else you've changed, or an entirely unrelated issue in the physical world (like something standing in the path of the signal, or the AP malfunctioning somehow), but it makes sense that s-n-w-o is not proceeding, even with `--any`, if the Wi-Fi network is not actually connected (the logs above tell exactly that).

I will try once more on my test device here, but could you please make sure that any physical/link layer Wi-Fi connectivity issues are excluded?

mdevaev, 2025-07-15:
@xuyenvuong sup?

xuyenvuong, 2025-07-16:
@mdevaev - I will get to the test this week and post my update. Sorry, I was OOT since your response, and didn't have the Pi with me.

xuyenvuong, 2025-07-16:
Here is my test:

First, without changing anything, my Pi's WiFi connection is working as expected. Everything was working with `--ignore=eth0` in the config.

Proceed with incremental changes:

Step 1.  

- Edit: `vi /etc/systemd/system/systemd-networkd-wait-online.service.d/override.conf` with `ExecStart=/usr/lib/systemd/systemd-networkd-wait-online --any`

Step 2. 

- Save file, `ro` & `reboot`

Observation after boot-up:

- `journalctl -f -b -n 200` has no error right after boot.
- WiFi connection is working.
- System was up without any delays and with mount point correctly display all files on the UI

What have I done differently this time? Apparently, for the last test, I believed I did both changes for `--any` and `_netdev` (fstab) for both config files. That could have been a culprit.

This bug can be closed. Thank you all for your help.

mdevaev, 2025-07-16:
We will add this fix to the distro, so you don't need to patch it yourself.

closed 2025-07-16 by mdevaev

mdevaev, 2025-07-16:
Released.