pikvm/pikvm #1525: No-op for redfish system patch and return boot override information

EdwardCooke, 2025-05-19:
**Is your feature request related to a problem? Please describe.**
When installing OKD (Kubernetes) using installer provisioned infrastructure it expects 2 fields to be returned and the patch request to return a 204.

**Describe the solution you'd like**
When using Redfish the patch to the system is a no-op and the necessary boot override parameters are returned.

**Describe alternatives you've considered**
Modifying the redfish.py file on every pi to include necessary changes.

**Additional context**
I understand that pikvm can't know or update bios settings to override boot order, that's pretty much a given. But doing a no-op and setting the boot order manually works fine for my use case and allows the installer to proceed.

I have already done this on my pikvm.

I can submit a PR with the changes if requested.

Using this as the get response:
```python
        return make_json_response({
            "@odata.id": "/redfish/v1/Systems/0",
            "@odata.type": "#ComputerSystem.v1_10_0.ComputerSystem",
            "Actions": {
                "#ComputerSystem.Reset": {
                    "ResetType@Redfish.AllowableValues": list(self.__actions),
                    "target": "/redfish/v1/Systems/0/Actions/ComputerSystem.Reset"
                },
                "#ComputerSystem.SetDefaultBootOrder": {
                    "target": "/redfish/v1/Systems/0/Actions/ComputerSystem.SetDefaultBootOrder"
                }
            },
            "Id": "0",
            "HostName": host,
            "PowerState": ("On" if atx_state["leds"]["power"] else "Off"),  # type: ignore
            "Boot": {
                "BootSourceOverrideEnabled": "Disabled",
                "BootSourceOverrideTarget": None,
            },
        }, wrap_result=False)
```

And this as the patch method
```python
    @exposed_http("PATCH", "/redfish/v1/Systems/0")
    async def __patch_handler(self, req: Request) -> Response:
        return Response(body=None, status=204)
```

The above changes work great for OKD.
```

mdevaev, 2025-05-20:
Sure, why not. Released in 4.73, please update OS with `pikvm-update`.

closed 2025-05-20 by mdevaev

EdwardCooke, 2025-05-20:
Thanks! Very much appreciated.