pikvm/pikvm #1530: Feature Request: Support for non-US keyboard layouts in virtual keyboard

devildant, 2025-05-25:
The virtual keyboard in PiKVM (HTML) only works with a US QWERTY layout. When controlling a target system with a different layout (e.g. French AZERTY), the keys don’t match, making it very hard to type correctly.

Request:
Please add support for selecting other keyboard layouts (like fr, de, es) in the virtual keyboard, so the key mapping matches the target system.

devildant, 2025-05-28:
Chrome Remote Desktop for exemple :p

russdan, 2025-08-06:
+1. Doesn't work for `en-gb` also, meaning using characters such as # and @ in passwords are a nightmare! Have tried editing `/etc/kvmd/override.yaml` to change the keymap settings but it doesn't seem to have any effect on the virtual keyboard (so when accessing via Android via browser).  Please provide a mechanism to configure the html keyboard mappings...

mdevaev, 2025-10-26:
I've added this ticket to my plan. It duplicates an existing old issue, so please subscribe to that issue and I will solve the problem there, and I will close this one as a duplicate.

Please follow here: #611

closed 2025-10-26 by mdevaev