pikvm/pikvm #1536: I want to send shortcut keys through the command line of SSH

worm001, 2025-05-31:
Discovered kvm_input.sh in the/etc/kvmd path, and after analyzing the content, obtained: [curl -X POST -k -u admin:$password "https://$IP/api/hid/events/send_key?key=$HOTKEY"] I built the command myself: [curl - X POST - k - u admin: admin“ https://192.168.1.111/api/hid/events/send_key?key=ControlLeft +AltLeft+Delete"] will generate an error message:
{
"ok": false,
"result": {
"error": "ValidatorError",
"error_msg": "The argument 'ControlLeft AltLeft Delete' is not a valid Keyboard key"
}
}
Do you want to know how to build a command if you want to send shortcut keys through the command line? Is there a specific command or script or program to implement it?

mdevaev, 2025-05-31:
Hello. Use `ControlLeft,AltLeft,Delete`.

https://docs.pikvm.org/api/

worm001, 2025-05-31:
It seems that it still doesn't work, the same prompt, I don't know where it went wrong.

mdevaev, 2025-05-31:
Hm, you're right, I thought I added this API.

You can use Python to send events via websocket:

https://docs.pikvm.org/api/#Sending%20key%20events%20using%20Python

worm001, 2025-06-01:
[image]    There are no error prompts for single key sending now, but the controlled system has no response. Shouldn't we use curl instead? Can only be programmed in Python?

worm001, 2025-06-01:
By studying the hid. py file, it is currently possible to achieve the combination of ctrl+alt+del using these 6 lines of commands:

curl -X POST -k -u admin:admin " https://localhost/api/hid/events/send_key?key=ControlLeft&state=true "
curl -X POST -k -u admin:admin " https://localhost/api/hid/events/send_key?key=AltLeft&state=true "
curl -X POST -k -u admin:admin " https://localhost/api/hid/events/send_key?key=Delete&state=true "
curl -X POST -k -u admin:admin " https://localhost/api/hid/events/send_key?key=Delete&state=false "
curl -X POST -k -u admin:admin " https://localhost/api/hid/events/send_key?key=AltLeft&state=false "
curl -X POST -k -u admin:admin " https://localhost/api/hid/events/send_key?key=ControlLeft&state=false "

But it feels a bit cumbersome, I don't know if it can be implemented with one or two commands.

worm001, 2025-06-01:
I researched it myself again and found that it can only send single key values. Code simplification:

curl -X POST -k -u admin:admin " https://localhost/api/hid/events/send_key?key=ControlLeft&state=true "
curl -X POST -k -u admin:admin " https://localhost/api/hid/events/send_key?key=AltLeft&state=true "
curl -X POST -k -u admin:admin " https://localhost/api/hid/events/send_key?key=Delete "

mdevaev, 2025-06-01:
I made a specal API for you in the new release. Update with `pikvm-update`.

```
curl -X POST -k -u admin:admin 'https://localhost/api/hid/events/send_shortcut?keys=ControlLeft,AltLeft,Delete'
```

closed 2025-06-01 by mdevaev