pikvm/pikvm #1539: kernel 6.6.45-13-rpi is broken

vbooka1, 2025-06-06:
**Describe the bug**
After the last update my piKVM randomly hangs some time after the reboot. The embedded WiFi network disappears, SSH closes connection, HTTPS hangs. Strangely HTTP works and this makes me think that something was broken with the encryption code in the kernel.

SSH:
```
$ telnet 192.168.111.222 22
Trying 192.168.111.222...
Connected to 192.168.111.222.
Escape character is '^]'.
Connection closed by foreign host.
```

HTTPS:

```
$ curl -v 192.168.111.222 -L
*   Trying 192.168.111.222:80...
* Connected to 192.168.111.222 (192.168.111.222) port 80
> GET / HTTP/1.1
> Host: 192.168.111.222
> User-Agent: curl/8.6.0
> Accept: */*
> 
< HTTP/1.1 301 Moved Permanently
< Server: nginx/1.27.5
< Date: Fri, 06 Jun 2025 04:18:30 GMT
< Content-Type: text/html
< Content-Length: 169
< Connection: keep-alive
< Location: https://192.168.111.222/
< 
* Ignoring the response-body
* Connection #0 to host 192.168.111.222 left intact
* Clear auth, redirects to port from 80 to 443
* Issue another request to this URL: 'https://192.168.111.222/'
*   Trying 192.168.111.222:443...
* Connected to 192.168.111.222 (192.168.111.222) port 443
* ALPN: curl offers h2,http/1.1
* TLSv1.3 (OUT), TLS handshake, Client hello (1):

(nothing happens here)
```

strangely HTTP works:

```
> curl -v 192.168.111.222
*   Trying 192.168.111.222:80...
* Connected to 192.168.111.222 (192.168.111.222) port 80
> GET / HTTP/1.1
> Host: 192.168.111.222
> User-Agent: curl/8.6.0
> Accept: */*
> 
< HTTP/1.1 301 Moved Permanently
< Server: nginx/1.27.5
< Date: Fri, 06 Jun 2025 04:18:28 GMT
< Content-Type: text/html
< Content-Length: 169
< Connection: keep-alive
< Location: https://192.168.111.222/
< 
<html>
<head><title>301 Moved Permanently</title></head>
<body>
<center><h1>301 Moved Permanently</h1></center>
<hr><center>nginx/1.27.5</center>
</body>
</html>
```

**To Reproduce**
Steps to reproduce the behavior, like:
1. run `pacman -Syu` (or perhaps `pikvm-update`, I usually update with `pacman`)
2. reboot
3. connect to embedded WiFi
4. wait until the WiFi randomly disconnects.

**Expected behavior**
I do not expect the above behavior to happen.

**Screenshots**
not applicable

**Desktop (please complete the following information):**
not applicable

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4] 
  raspberry Pi 4 model B, 4GB
  - PiKVM platform [e.g. v2-hdmi] 
  pikvm-v3-hat
  - Video capture type [e.g. CSI bridge] 
  not applicable
  - KVMD version: `pacman -Q | grep kvmd`
  kvmd 4.82-1
kvmd-fan 0.32-1
kvmd-platform-v3-hdmi-rpi4 4.82-1
kvmd-webterm 0.50-1
  - uStreamer version: `pacman -Q | grep ustreamer`
  ustreamer 6.37-1
  - Linux kernel: `uname -a`
Linux pikvm 6.6.45-13-rpi #1 SMP Tue May 27 17:16:41 UTC 2025 armv7l GNU/Linux

**Additional context**
I have just realized that I may use a standard Arch repo not a piKVM repo, how do I verify?

```
# cat /etc/pacman.d/mirrorlist        
Server = http://mirror.archlinuxarm.org/$arch/$repo
```

vbooka1, 2025-06-06:
I have changed the journald storage to Persistent, hope it will show what happens before the hang. Anything else I coud do to debug this?

mdevaev, 2025-06-06:
Please try to reflash the service if possible, update and check again.

Best regards,
Max Devaev

vbooka1, 2025-06-06:
I have found another problem: now `ping -c 1` has just 10ms timeout so on a slow mobile network (I use a USB modem to have piKVM always online) it always fails:

```
[root@pikvm ~]# ping -c 1 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms

[root@pikvm ~]# echo $?
1
```
strace shows this:

```
setitimer(ITIMER_REAL, {it_interval={tv_sec=0, tv_usec=0}, it_value={tv_sec=0, tv_usec=10000}}, NULL) = 0 <0.000042>
```

If I explicitly tell ping to wait for at least 1 second then ping works well:

```
[root@pikvm ~]# ping -c 1 1.1.1.1 -W 1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=56 time=35.6 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 35.581/35.581/35.581/0.000 ms
[root@pikvm ~]# echo $?
0
```

I can't recall for sure but I think the ping timeout was higher before the update. And I believe it should be higher, not just 10ms. Should I open a separate bug report for ping issue?

mdevaev, 2025-06-06:
> Hi, you mean reflash the whole SD card?

Yes. If you're using PiKVM V3 with 32-bit kernel now, you need this image: https://files.pikvm.org/images/v3-hdmi-rpi4/arm/v3-hdmi-rpi4-box-latest.img.xz

> Should I open a separate bug report for ping issue?

Let's check the kernel for now. I think this is some kind of local problem you have. If the error with the kernel is reproduced, then we will deal with it further.

mdevaev, 2025-06-09:
No feedback and we can't reproduce it on our side. I can recommend you to reflash the OS just to be sure.

closed 2025-06-09 by mdevaev

mdevaev, 2025-07-15:
@vbooka1 Glad to hear it. Replace the memory card just in case.