pikvm/pikvm #1543: [ pikvm-update ] : "directory permissions differ" error messages

RastaTaz, 2025-06-07:
**Describe the bug**
Error message on pikvm-update command output refering to directory permissions differing from what was expected.

**To Reproduce**
1. Run 'pikvl-update' command
2. Depending on packages to update, you may get errors (see mine in screenshot section) refing to directory permission differing from what is expected.

**Expected behavior**
No error on package update deployments, especially for system packages which i did not modify/tune.

**Screenshots**
Here are the errors I got, sorry I did not take the whole output or note that packages, I did just extract errors themselves:
```
warning: directory permissions differ on /usr/share/polkit-1/rules.d/
filesystem: 750  package: 755
warning: directory permissions differ on /var/log/journal/
filesystem: 755  package: 2755
warning: directory permissions differ on /var/lib/kvmd/pst/
filesystem: 1775  package: 755
```

**Desktop (please complete the following information):**
 - OS: PiKVM for Rpi 4

**PiKVM info:**
  - Raspberry Pi board version: RPi 4
  - PiKVM platform: don't know whete to get this information and can't recall...
  - Video capture type: same as above, but does not see to be relevant to issue
  - KVMD version: `pacman -Q | grep kvmd`
```
kvmd 4.82-1
kvmd-fan 0.32-1
kvmd-platform-v3-hdmi-rpi4 4.82-1
kvmd-webterm 0.50-1
```
  - uStreamer version: `pacman -Q | grep ustreamer`
``` ustreamer 6.37-1```
  - Linux kernel: `uname -a`
```Linux pikvm 6.6.45-13-rpi #1 SMP Tue May 27 17:16:41 UTC 2025 armv7l GNU/Linux```

**Additional context**
-

Just one question: should I hand-correct those by setting expected permissions according to error messages ?

RastaTaz, 2025-06-07:
I should correct initial statements: those are not error messages but warning messages...

closed 2025-06-29 by mdevaev

RastaTaz, 2025-06-29:
Thank @mdevaev 

Just to let you know: on update I performed today, I had similar warnings on systemd related folders (see pikvm-update output extract below).

```
(51/80) upgrading systemd                            [############################] 100%
warning: directory permissions differ on /usr/share/polkit-1/rules.d/
filesystem: 750  package: 755
warning: directory permissions differ on /var/log/journal/
filesystem: 755  package: 2755
```

mdevaev, 2025-06-29:
It's not a problem, you can ignore it. But if you want, you can fix it manually:
```
rw
chmod 755 /usr/share/polkit-1/rules.d/
rm -rf /var/log/journal/
ro
```
The `/var/log/journal/` directory should not exist normally on PiKVM.